retro-esp32/RetroESP32 #1: START + SELECT Hotkey Saves, but doesn't exit emulator

32teeth, 2019-05-21:
**Is your feature request related to a problem? Please describe.**
START + SELECT now saves the ROM in emulator, but hangs on hourglass screen
Required power up/down to get back to launcher

**Describe the solution you'd like**
START + SELECT should save ROM and exit back to Menu

**Describe alternatives you've considered**
N/A

**Additional context**
N/A

pebri86, 2019-05-23:
Hi, currently i don't have hardware to test, but looking at the code it should be work as expected. When Start + Select button pressed shortly, it emulates the menu button and trigger do menu home behavior, and then it should exiting audio video task and create .sav state file -> set boot to menu (ota 0) -> reboot esp.

Is .sav file created successfully? if yes then esp doesn't restart for something reason.
Could you post the serial logs?

32teeth, 2019-05-24:
### First Boot
> Select Game in NES Emulator
Runs 'Blades Of Steel' in this example

<details><summary>Click to Expand</summary>
<p>

```shell
MONITOR
--- idf_monitor on /dev/cu.SLAB_USBtoUART 115200 ---
--- Quit: Ctrl+] | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
ets Jun 8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5828
load:0x40078000,len:9020
load:0x40080000,len:6120
0x40080000: _iram_start at /Users/eugene/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

entry 0x40080334
0x40080334: _KernelExceptionVector at ??:?

I (28) boot: ESP-IDF v3.1-odroid-1-g08f2c8f26 2nd stage bootloader
I (28) boot: compile time 21:04:48
I (36) boot: Enabling RNG early entropy source...
I (36) boot: SPI Speed : 80MHz
I (38) boot: SPI Mode : DIO
I (42) boot: SPI Flash Size : 16MB
I (47) boot: Partition Table:
I (50) boot: ## Label Usage Type ST Offset Length
I (57) boot: 0 nvs WiFi data 01 02 00009000 00004000
I (65) boot: 1 otadata OTA data 01 00 0000d000 00002000
I (72) boot: 2 phy_init RF data 01 01 0000f000 00001000
I (80) boot: 3 factory factory app 00 00 00010000 000f0000
I (87) boot: 4 storage Unknown data 01 82 00100000 00100000
I (95) boot: 5 app_0 OTA app 00 10 00200000 00100000
I (102) boot: 6 app_1 OTA app 00 11 00300000 00100000
I (110) boot: 7 app_2 OTA app 00 12 00400000 00100000
I (117) boot: 8 app_3 OTA app 00 13 00500000 00100000
I (125) boot: 9 data_0 unknown 40 00 00600000 00400000
I (133) boot: End of partition table
I (137) boot: Defaulting to factory image
I (141) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x2cd10 (183568) map
I (204) esp_image: segment 1: paddr=0x0003cd38 vaddr=0x3ffb0000 size=0x0316c ( 12652) load
I (208) esp_image: segment 2: paddr=0x0003feac vaddr=0x3ffb316c size=0x00000 ( 0) load
I (211) esp_image: segment 3: paddr=0x0003feb4 vaddr=0x40080000 size=0x0015c ( 348) load
0x40080000: _iram_start at /Users/eugene/esp/esp-idf/components/freertos/xtensa_vectors.S:1685

I (220) esp_image: segment 4: paddr=0x00040018 vaddr=0x400d0018 size=0x29f64 (171876) map
0x400d0018: _flash_cache_start at ??:?

I (278) esp_image: segment 5: paddr=0x00069f84 vaddr=0x4008015c size=0x002a4 ( 676) load
0x4008015c: _WindowUnderflow12 at /Users/eugene/esp/esp-idf/components/freertos/xtensa_vectors.S:1876

I (279) esp_image: segment 6: paddr=0x0006a230 vaddr=0x40080400 size=0x09a4c ( 39500) load
I (299) esp_image: segment 7: paddr=0x00073c84 vaddr=0x400c0000 size=0x00000 ( 0) load
I (299) esp_image: segment 8: paddr=0x00073c8c vaddr=0x50000000 size=0x00000 ( 0) load
I (312) boot: Loaded app from partition at offset 0x10000
I (312) boot: Disabling RNG early entropy source...
I (318) cpu_start: Pro cpu up.
I (321) cpu_start: Starting app cpu, entry point is 0x40080fc8
0x40080fc8: Cache_Read_Enable at /Users/eugene/esp/esp-idf/components/esp32/include/rom/cache.h:181 (discriminator 1)
(inlined by) call_start_cpu0 at /Users/eugene/esp/esp-idf/components/esp32/cpu_start.c:169 (discriminator 1)

I (0) cpu_start: App cpu up.
I (332) heap_init: Initializing. RAM available for dynamic allocation:
[... 820 lines cut ...]

0x400d1ff0: esp_task_wdt_init at /Users/eugene/esp/esp-idf/components/esp32/task_wdt.c:236

Entering gdb stub now.
$T0b#e6GNU gdb (crosstool-NG crosstool-ng-1.22.0-80-g6c4433a) 7.10
Copyright (C) 2015 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later 
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law. Type "show copying"
and "show warranty" for details.
This GDB was configured as "--host=x86_64-build_apple-darwin16.3.0 --target=xtensa-esp32-elf".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
.
Find the GDB manual and other documentation resources online at:
.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /Users/eugene/Desktop/github/GabozeExpress/Public/Software/odroid-go-firmware/build/odroid-go-firmware.elf...done.
Remote debugging using /dev/cu.SLAB_USBtoUART
0x40088243 in ?? ()
(gdb)
```
</p>
</details>

32teeth, 2019-05-27:
@pebri86 
@gabozepocaio 

### First Boot
> Select Game in NES Emulator
Runs 'Blades Of Steel' in this example

<details><summary>Click to Expand</summary>
<p>
```shell
rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:7156
load:0x40078000,len:10320
load:0x40080400,len:7060
entry 0x40080754
I (29) boot: ESP-IDF v3.2-dev-1706-g24e4887c7-dirty 2nd stage bootloader
I (29) boot: compile time 17:39:53
I (40) boot: Enabling RNG early entropy source...
I (40) qio_mode: Enabling QIO for flash chip GD
I (41) boot: SPI Speed      : 80MHz
I (45) boot: SPI Mode       : QIO
I (49) boot: SPI Flash Size : 16MB
I (53) boot: Partition Table:
I (57) boot: ## Label            Usage          Type ST Offset   Length
I (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (72) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (79) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (86) boot:  3 firmware         factory app      00 00 00010000 000f0000
I (94) boot:  4 gogo             OTA app          00 10 00100000 00100000
I (101) boot:  5 nesemu-go        OTA app          00 11 00200000 00100000
I (109) boot:  6 gnuboy-go        OTA app          00 12 00300000 00100000
I (116) boot:  7 smsplusgx-go     OTA app          00 13 00400000 00200000
I (124) boot: End of partition table
I (2128) boot: Detect a condition of the factory reset
I (2128) boot: Data partitions to erase: nvs
I (2128) boot_comm: ## Label            Usage Offset   Length   Cleaned
I (2266) boot_comm:  0 nvs              data  00009000 00004000 [yes]
I (2358) boot_comm:  1 otadata          data  0000d000 00002000 [yes]
I (2358) boot_comm:  2 phy_init         data  0000f000 00001000 [no]
I (2361) boot: Defaulting to factory image
I (2365) esp_image: segment 0: paddr=0x00010020 vaddr=0x3f400020 size=0x0f8dc ( 63708) map
I (2391) esp_image: segment 1: paddr=0x0001f904 vaddr=0x3ff80000 size=0x00000 (     0) load
I (2391) esp_image: segment 2: paddr=0x0001f90c vaddr=0x3ff80000 size=0x00000 (     0) load
I (2398) esp_image: segment 3: paddr=0x0001f914 vaddr=0x3ffb0000 size=0x006fc (  1788) load
I (2407) esp_image: segment 4: paddr=0x00020018 vaddr=0x400d0018 size=0x23ec8 (147144) map
0x400d0018: _flash_cache_start at ??:?
```
</p>
</details>

closed 2019-06-13 by 32teeth