retro-esp32/RetroESP32 #3: SD Card Issue

32teeth, 2019-05-28:
**Describe the bug**
SD Crashing in both GoGo and Go-Play with different errors

**To Reproduce**
Steps to reproduce the behavior:
1. Run ```ota``` script
2. Select a game
3. Hold Select+Start to save and restart
4. On restart behaviour shows sd card image with exclamation
5. Multiple SD Cards where tried, same error

**Expected behavior**
1. Save State
2. Restart GoGo Launcher into menu

###mAdditional context
> After Reset

<details><summary>Click to Expand</summary>
<p>

```shell
I (411) cpu_start: App cpu up.
I (1301) spiram: SPI SRAM memory test OK
I (1301) heap_init: Initializing. RAM available for dynamic allocation:
I (1302) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (1308) heap_init: At 3FFC7E28 len 000181D8 (96 KiB): DRAM
I (1314) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (1320) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (1327) heap_init: At 40091800 len 0000E800 (58 KiB): IRAM
I (1333) cpu_start: Pro cpu start user code
I (1338) spiram: Adding pool of 4096K of external SPI memory to heap allocator
I (17) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
I (18) spiram: Reserving pool of 32K of internal memory for DMA/internal allocations
odroid start.
odroid_input_gamepad_init done.
app_main: Non deep sleep startup
ili9341_init: line_buffer_put(0x3ffd4ed8)
spi_task: Entered.
ili9341_init: line_buffer_put(0x3ffd5b5c)
LCD: calling ili_init.
LCD: calling backlight_init.
LCD Initialized (40000000 Hz).
ADC: Characterized using eFuse Vref
E (468) spi_master: spi_bus_initialize(242): host already in use
I (468) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (518) sdmmc_sd: sdmmc_init_sd_if_cond: send_if_cond (1) returned 0x108
E (518) spi_master: spi_bus_free(359): not all CSses freed
==========================
GoGo
==========================
odroid_sdcard_open: esp_vfs_fat_sdmmc_mount failed (264)
==========================
I (528) I2S: DMA Malloc info, datalen=blocksize=2048, dma_buf_count=4
I (538) I2S: PLL_D2: Req RATE: 16000, real rate: 1004.000, BITS: 16, CLKM: 83, BCK: 60, MCLK: 83.333, SCLK: 32128.000000, diva: 64, divb: 21
odroid_settings_RomFilePathGet: value='/sd/roms/gbc/Legend of Zelda, The - Link's Awakening DX (U) (V1.2) (C).gbc'
extension=gbc
resume - extension=gbc, slot=2
I (708) esp_image: segment 0: paddr=0x00400020 vaddr=0x3f400020 size=0x6192c (399660) map
I (948) esp_image: segment 1: paddr=0x00461954 vaddr=0x3ffb0000 size=0x03340 ( 13120)
I (958) esp_image: segment 2: paddr=0x00464c9c vaddr=0x3ffb3340 size=0x00000 (     0)
I (958) esp_image: segment 3: paddr=0x00464ca4 vaddr=0x40080000 size=0x00400 (  1024)
0x40080000: _WindowOverflow4 at /Users/eugene/Desktop/github/GabozeExpress/Public/Software/Tools/esp-idf/components/freertos/xtensa_vectors.S:1685

I (978) esp_image: segment 4: paddr=0x004650ac vaddr=0x40080400 size=0x0af64 ( 44900)
I (1008) esp_image: segment 5: paddr=0x00470018 vaddr=0x400d0018 size=0x29a40 (170560) map
0x400d0018: _flash_cache_start at ??:?

I (1108) esp_image: segment 6: paddr=0x00499a60 vaddr=0x4008b364 size=0x06c6c ( 27756)
I (1128) esp_image: segment 7: paddr=0x004a06d4 vaddr=0x400c0000 size=0x00064 (   100)
I (1128) esp_image: segment 8: paddr=0x004a0740 vaddr=0x50000000 size=0x00000 (     0)
I (1138) esp_image: segment 9: paddr=0x004a0748 vaddr=0x50000000 size=0x00000 (     0)
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x33 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:6464
ho 0 tail 12 room 4
load:0x40078000,len:10180
load:0x40080400,len:6692
entry 0x40080764
I (31) boot: ESP-IDF v3.2-dev-1707-g08db568ba-dirty 2nd stage bootloader
I (31) boot: compile time 07:41:56
I (40) boot: Enabling RNG early entropy source...
I (40) boot: SPI Speed      : 40MHz
I (42) boot: SPI Mode       : DIO
I (46) boot: SPI Flash Size : 16MB
I (50) boot: Partition Table:
I (53) boot: ## Label            Usage          Type ST Offset   Length
I (61) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (68) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (76) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (83) boot:  3 factory          factory app      00 00 00010000 000f0000
I (91) boot:  4 storage          Unknown data     01 82 00100000 00100000
I (98) boot:  5 app_0            OTA app          00 10 00200000 00100000
I (105) boot:  6 app_1            OTA app          00 11 00300000 00100000
I (113) boot:  7 app_2            OTA app          00 12 00400000 00100000
I (121) boot:  8 app_3            OTA app          00 13 00500000 00100000
I (128) boot:  9 data_0           unknown          40 00 00600000 00400000
I (136) boot: End of partition table
I (140) esp_image: segment 0: paddr=0x00400020 vaddr=0x3f400020 size=0x6192c (399660) map
I (289) esp_image: segment 1: paddr=0x00461954 vaddr=0x3ffb0000 size=0x03340 ( 13120) load
I (294) esp_image: segment 2: paddr=0x00464c9c vaddr=0x3ffb3340 size=0x00000 (     0) load
I (296) esp_image: segment 3: paddr=0x00464ca4 vaddr=0x40080000 size=0x00400 (  1024) load
0x40080000: _WindowOverflow4 at /Users/eugene/Desktop/github/GabozeExpress/Public/Software/Tools/esp-idf/components/freertos/xtensa_vectors.S:1685

I (305) esp_image: segment 4: paddr=0x004650ac vaddr=0x40080400 size=0x0af64 ( 44900) load
I (332) esp_image: segment 5: paddr=0x00470018 vaddr=0x400d0018 size=0x29a40 (170560) map
0x400d0018: _flash_cache_start at ??:?

I (392) esp_image: segment 6: paddr=0x00499a60 vaddr=0x4008b364 size=0x06c6c ( 27756) load
I (404) esp_image: segment 7: paddr=0x004a06d4 vaddr=0x400c0000 size=0x00064 (   100) load
I (404) esp_image: segment 8: paddr=0x004a0740 vaddr=0x50000000 size=0x00000 (     0) load
I (411) esp_image: segment 9: paddr=0x004a0748 vaddr=0x50000000 size=0x00000 (     0) load
I (430) boot: Loaded app from partition at offset 0x400000
I (430) boot: Disabling RNG early entropy source...
I (432) spiram: Found 64MBit SPI RAM device
I (436) spiram: SPI RAM mode: flash 80m sram 80m
I (442) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (449) cpu_start: Pro cpu up.
I (452) cpu_start: Starting app cpu, entry point is 0x4008140c
0x4008140c: esp_timer_start_periodic at /Users/eugene/Desktop/github/GabozeExpress/Public/Software/Tools/esp-idf/components/esp32/esp_timer.c:176

I (444) cpu_start: App cpu up.
I (463) heap_init: Initializing. RAM available for dynamic allocation:
I (470) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (476) heap_init: At 3FFC1738 len 0001E8C8 (122 KiB): DRAM
I (482) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (489) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (495) heap_init: At 40091FD0 len 0000E030 (56 KiB): IRAM
I (501) cpu_start: Pro cpu start user code
I (71) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
gnuboy (20190528-c573693f91).
odroid_input_gamepad_init done.
app_main: Non deep sleep startup
ili9341_init: line_buffer_put(0x3ffc6b78)
spi_task: Entered.
ili9341_init: line_buffer_put(0x3ffc77fc)
LCD: calling ili_init.
LCD: calling backlight_init.
LCD Initialized (40000000 Hz).
odroid_settings_RomFilePathGet: value='/sd/roms/gbc/Legend of Zelda, The - Link's Awakening DX (U) (V1.2) (C).gbc'
loader: Reading from sdcard.
E (958) spi_master: spi_bus_initialize(242): host already in use
I (968) gpio: GPIO[22]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0
E (978) sdmmc_io: sdmmc_io_reset: unexpected return: 0x109
E (978) spi_master: spi_bus_free(359): not all CSses freed
==========================
Go Play
==========================
odroid_sdcard_open: esp_vfs_fat_sdmmc_mount failed (265)
==========================
abort() was called at PC 0x400f4693 on core 0

Backtrace: 0x4008f9eb:0x3ffc3820 0x4008fcb9:0x3ffc3840 0x400f4693:0x3ffc3860 0x400f48d3:0x3ffc3880 0x400d3600:0x3ffc38a0 0x400d1a78:0x3ffc3900
0x400d3600: write_partition_table at /Users/eugene/Desktop/github/GabozeExpress/Public/Software/Odroid/odroid-go-firmware/main/main.c:357
 (inlined by) flash_firmware at /Users/eugene/Desktop/github/GabozeExpress/Public/Software/Odroid/odroid-go-firmware/main/main.c:902

0x400d1a78: esp_intr_alloc_intrstatus at /Users/eugene/Desktop/github/GabozeExpress/Public/Software/Tools/esp-idf/components/esp32/intr_alloc.c:548
```
</p>
</details>

pebri86, 2019-05-29:
Hi,
looking at the logs it seems you're using unpatched esp-idf version. To shared spi lines between tft and sdcard you must use eap-idf forked by crashoverride.

based on the logs you're use esp-idf v3.2-dev branch. Please switch to v3.2-odroid branch. then recompile all the source including emulators.

32teeth, 2019-06-13:
## Resolved
> Switch Branches to V3.2

[image]
[image]
[image]

closed 2019-06-13 by 32teeth