retro-esp32/RetroESP32 #7: Crash on first try with sub folders

Voljega, 2019-07-19:
Just downloaded firmware + skeleton sd card and burned the firmware

On first boot I chose my theme color and then went on NES system at the top.
I was told we could use subfolders in this firmware, so I just copied my setup from a retrogaming distrib for the four main system :

[image]

With this subfolders, no games are displayed for one system, just one system has one game displayed. But Each system contains the whole romset in these subfolders ...
like this :

[image]

Anyway nes was saying 'no games found' I clicked A anyway, then something like 'Run' appeared, I clicked A again and the emulator seemed to launch, but I ended up with the white screen and red crossed sd card icon, the one saying ' - Files are not properly written to the SD card, or its contents are corrupted.' (https://wiki.odroid.com/odroid_go/emulator/usage_go_play)

And now I can't do anything, Force exit emulator (menu hold when powering on) does nothing, nor menu, nor reset, seems the firmware is totally stuck

so TLDR :
- subfolders doesn't to be supported at all
- when the emulator crashes there doesn't seem to be anyway to go back to menu
- strange command run in emulator with no game

32teeth, 2019-07-19:
I like this feature alot.
We will implement in next few days

32teeth, 2019-07-19:
Closing this one and creating two separate Issues

- [ ] [Ability to Navigate Sub Folders in ROM Directories](https://github.com/retro-esp32/RetroESP32/issues/8)
- [ ] [Emulator Crashes if Target File is Blank](https://github.com/retro-esp32/RetroESP32/issues/9)

closed 2019-07-19 by 32teeth

Voljega, 2019-07-19:
OK but in the meantime is there anyway to come back to menu or will I need to reflash the firmware ?

32teeth, 2019-07-19:
All emulators have a Wake Up interrupt

```
            if (bootState.values[ODROID_INPUT_MENU])
            {
                // Force return to menu to recover from
                // ROM loading crashes

                // Set menu application
                odroid_system_application_set(0);

                // Reset
                esp_restart();
            }
```

You can power up and hold **Menu** (Start + Select) to go back to Retro ESP
Or...  ...power up and hold **B** to go back to Odroid Firmware (Unless you used ```./ota``` script)

Voljega, 2019-07-19:
OK it worked with start + select

AmitMamoria, 2020-09-16:
Hi can someone please tell me how to run retro arch launcher or how to run games from sd card. I am willing to make my own retro console from scratch ie. using with individual parts  instead of buying pre-made boards that are available on google.
I have googled and found some requirements for making retro console. 

I have Following parts with me.
1.) Esp32 Dev Kit Board ( https://www.electronicscomp.com/esp32-development-board-with-wifi-bluetooth-india?gclid=CjwKCAjw74b7BRA_EiwAF8yHFE4jsQzOF9uV9N4Jpo5NEkk4oYJbGYL4lAK6atGl-U6_IuWaywzbmxoCOF0QAvD_BwE)
2.) TFT LCD
3.) SD Card Reader.

 Please Help !!

fizarum, 2020-09-16:
Hi, first of all you can check this repo: https://github.com/airainchoc/RetroESP32/tree/Hardware and this BOM: https://github.com/airainchoc/RetroESP32/blob/Hardware/bom/RetroESP32.csv
make sure your board has this esp32 module: ESP32-WROVER (16MB) there is some picture of this soc: https://ae01.alicdn.com/kf/Hf494a4a947114e01b17cf46851904a443.jpg_q50.jpg
so it looks different from Esp32 Dev Kit Board