retro-esp32/RetroESP32 #14: UI crashes if there are a lot of ROMs

tmihai20, 2019-07-24:
I have hundreds of ROM files on my microSD card and when I navigate to GameBoy Color ROMS launcher is restarted and it goes back to the first list of available ROMS. I have a lot of ROMs because I want to have versions from all regions. This is happening if I am going to the SegaMasterSystem folder (navigating to the left).

Steps to reproduce the behavior:
1. Copy a lot of ROMs in all ROM folders (I have hundreds of them).
2. Navigate right to see all available ROMs.
3. In my case, when going to GameBoyColor ROMs, UI/launcher crashes and goes back to main screen. This is happening even I navigate left to SegaMasterSystem ROMs.
I am using the release tagged with a3c6d90.

32teeth, 2019-07-24:
@tmihai20 
> Approximate count of files?
- [ ] 50
- [ ] 100
- [ ] 200
- [ ] 500

@32teeth 
> This may be related to:
- [ ] sort_files
- [ ] quick_sort
Not being able to handle extreme amount of file names

Too much for memory to handle
```
    const int MAX_FILES = 1024;
    char** result = (char**)malloc(MAX_FILES * sizeof(void*));
```

tmihai20, 2019-07-24:
I have 303 col, 117 gb, 1213 gbc, 76 in gg, 846 in nes, and 1483 in sms. Some files have characters like these ones 76-in-1 (U) [p1][!]. The exclamation mark is present quite a lot as well. I did not rename them, I just copied them as I found them. I guess the main issue is with the GBC and SMS folders, there are more than 1000 files in them.

32teeth, 2019-07-25:
thanks for the response @tmihai20 

ill test to see what happens when we increase the MAX_FILES limit.
```
const int MAX_FILES = 1024;
char** result = (char**)malloc(MAX_FILES * sizeof(void*));
```

closed 2019-07-27 by 32teeth

tmihai20, 2019-07-29:
It looks like this issue is not yet fixed with 1.0.1 release. UI still crashes when I go to the emulator that has more than 1024 ROMs in it. Perhaps the issue was not only related to MAX_FILES.

tmihai20, 2019-07-30:
Maybe changing the sort algorithm would help (like using [Cycle Sort](http://en.wikipedia.org/wiki/Cycle_sort) or [Selection Sort](http://en.wikipedia.org/wiki/Selection_sort)).

esmith13, 2019-12-11:
I have this issue as a first time user on Odroid Go. I am using the current release, v1.6 and have under 1024 roms in each system (my largest is under 800 and the rest are all under 600). I have removed the following characters from all file names already:
&  ,  '  !
None of my roms are compressed. PCE and GG lists never show and cause an immediate crash. Aside from those two, moving from list to list a few times in a row always causes a crash as well, even on a list that works fine normally. Can't get to any other systems than NES, GB, GBC & SMS since it crashes when I get to GG going right as well as PCE when going left.

Attached is a full file & folder listing of my SD Card 'roms' folder.
[SD Card.txt](https://github.com/retro-esp32/RetroESP32/files/3951514/SD.Card.txt)