retro-esp32/RetroESP32 #15: UI/launcher does not show all the ROMs

tmihai20, 2019-07-24:
UI/launcher does not show all the ROMs in any of the emulators (NES and so on) when it does not crash, I have hundreds of them, it shows 70-80 ROMs

Steps to reproduce the behavior:
1. Copy a lot of ROMs (for instance, in the NES folder). It does not show all ROMs in any of the emulators.
2. Navigate down to select a ROM like Mortal Kombat.
3. UI/launcher only shows a few (70-80 ROMs)

I am using the release tagged with a3c6d90.

32teeth, 2019-07-24:
Thanks @tmihai20 
For clarity.
Do you have sub folders in your ```rom/[emulator folder]/``` folders?

tmihai20, 2019-07-24:
I do not have subfolders in the rom/[emulatorfolder].

32teeth, 2019-07-25:
@tmihai20 

my apologies for lack of clarity.
i meant

```
 // nes example (not real names)
 roms
 -- action
     -- [rom name].nes
     -- [rom name].nes
     -- [rom name].nes
     -- [rom name].nes
     -- [rom name].nes
 -- rpg
     -- [rom name].nes
     -- [rom name].nes
     -- [rom name].nes
     -- [rom name].nes
     -- [rom name].nes
 -- and so on
```

tmihai20, 2019-07-26:
No, the ROMs for a certain type of console are in a single folder. So the NES folder has only files with NES ROMs and so on. I am not arranging them by type or any other criteria.

tmihai20, 2019-07-29:
This issue is now fixed with the 1.0.1 release. I can see all the existing ROMs.

closed 2019-07-29 by tmihai20