retro-esp32/RetroESP32 #17: Remember last used/selected ROM in Launcher

tmihai20, 2019-07-29:
I would love to be able to go to the last ROM I selected in any of the emulator. Whenever I go to the list I get the first ROM in the list.

Whenever I go from one emulator to the other, Launcher should remember the last used or selected ROM in the list.

Right now I have to manually go to the ROM I want to, which most of the times is not the first one. Using the SELECT/START buttons to quickly navigate helps a bit, but I have to go through a lot of files to get to the one I want (like Mortal Kombat).

pebri86, 2019-07-30:
actually it does, partially. when you go back from emulator it remember last rom and emulator loaded, but not when you power on from hard switch. maybe we could implement some function like when press B it would try to load last emulator and roms.

tmihai20, 2019-07-30:
I have not tried that. The sort of official firmware does that even when powered off. I do not especially like the option to keep going back until it would load the last ROM, but people owning an oDroid GO are already using it, so it could be present here too.

pebri86, 2019-07-31:
The settings are already there, just need a bit change of code. But i don't have hardware now to test it

32teeth, 2019-08-06:
> I would love to be able to go to the last ROM I selected in any of the emulator. Whenever I go to the list I get the first ROM in the list.

Currently, if you are in an emulator and hit START+SELECT (MENU) and Exit Game, you will return to the last selected System and Game

> Whenever I go from one emulator to the other, Launcher should remember the last used or selected ROM in the list.

See Above

> Right now I have to manually go to the ROM I want to, which most of the times is not the first one. Using the SELECT/START buttons to quickly navigate helps a bit, but I have to go through a lot of files to get to the one I want (like Mortal Kombat).

If you shut down / power down from inside a Game/Emulator, that Emulator/Game Restarts on Power up.

If you exit the game, and return to the Launcher, perhaps we (@32teeth and @pebri86 ) can implement an in Launcher Menu (Like RetroPi) that has a Soft Shutdown?

32teeth, 2019-08-06:
@tmihai20 if you are okay, i will add this to the project board, but refine the language based on your response

tmihai20, 2019-08-06:
@32teeth yes, please add it to the project board, this needs to be known by all the users. You can close the issue as well.

32teeth, 2019-09-18:
In Project Board

closed 2019-09-18 by 32teeth