retro-esp32/RetroESP32 #18: External DAC

ripper121, 2019-08-15:
Please add an external DAC, the ESP32's build in DAC is to crappy to get a good sound out of it.
I would love to see it with external DAC, then I could listen to my mp3's <3

32teeth, 2019-08-16:
Thanks @ripper121 for the suggestion.

We are continuously exploring enhancement and PWM Audio (Standard on Odroid Go) to Stereo is one of them.

Currently we use an LM4875 from Texas Instruments

Some of the Digital Audio Convertors require Data and Clock as well as are power hungry

That said, we continue to research

closed 2019-09-18 by 32teeth

ripper121, 2019-09-19:
LM4875 is no DAC?! its a AMP...
Use the PCM5102a (DAC, needs no extra AMP) its great sound and compatible with all the Emulators.

ripper121, 2019-09-19:
With LM4875 you will amplify the crappy sound from the ESP32 build in DAC! :(