retro-esp32/RetroESP32 #19: Install and build questions

mogenson, 2019-09-03:
Hello, I appreciate this project that you're putting together. I'd like to use it to start hacking on my Odroid Go. However, I'm having a couple of problems getting setup and replicating the released Retro ESP32.fw.

First, I noticed after the install that I have some tar balls in my Downloads folder, new ~/Arduino and ~/esp directories, and python utilities for building sphinx documentation in ~/.local/bin. If a copy of the ESP-IDF is contained inside this repo in Tools/esp-idf, why are these other directories needed?

Could the installer be a little more polite and either download resources inside this repo or ask me before peppering files around my hard drive? An uninstall script would also be nice, so I know I haven't lost track of anything.

It looks like the install script also tries to download and bootstrap python pip. Could you check to see if pip is installed and prompt to install system-wide packages before executing a downloaded python file with root privileges?

Also, does this project require python 2, python 3, or a mix of both? The config for some components ask for the name of the python 2 executable but I'm not sure what the install script is trying to install. I believe all python installations should have a python2 or python3 symlink to the interpreter so you can be explicit.

Finally, I appreciate your choice of an interactive shell, but is there a reason why the Bash scripts mentioned in the README are Zsh scripts? It's unusual to make Zsh a dependency for a build script when Bash is everywhere.

I'm still tracking down where my build is failing. The clear commands in the scripts hide any printed errors and the build doesn't stop if one component failed. There's a few variables like $NAME that are unquoted and may be expanding into multiple whitespace separated words.

32teeth, 2019-09-05:
> Finally, I appreciate your choice of an interactive shell, but is there a reason why the Bash scripts mentioned in the README are Zsh scripts? It's unusual to make Zsh a dependency for a build script when Bash is everywhere.

@mogenson good point here
although ZSH is going to be native in OS X Catalina (October) i have reverted to Bash for Linux

mogenson, 2019-09-06:
After re-reading my post, I don't mean to be critical or negative. I've been hoping to see a little community build up around these retro ESP32 handhelds like the Odroid Go. This project seems like a very good place to start.

I've gotten my firmware to build. And I've proposed some fixes to the build scripts in #20.

32teeth, 2019-09-07:
> After re-reading my post, I don't mean to be critical or negative. 

No harm done, no offence take. ```#!/bin/zsh``` just sometimes lets me be lazier ;)

Did you try both ```./firmware``` and ```./ota```
What was missing from the ```./install``` process

closed 2019-09-07 by 32teeth