retro-esp32/RetroESP32 #22: Menu crashes trying to list files in Sega Master System and Sega Game Gear

shockdesign, 2019-09-13:
**Describe the bug**
When shifting right to view the games in the SMS column, it'll crash trying to load the list of games.

**To Reproduce**
Place some SMS roms into the correct directory (SMS).
With the SDCard back in the device, try to navigate to view the ROMs available for the Sega Master System. 
The menu crashes and starts again.

**Expected behavior**
Should expect a list of ROMs to appear.

**Device:**
ODROID GO

Is there a better way to debug this issue?

32teeth, 2019-09-13:
@shockdesign 
Can you maybe tell me how many games you have in thors folders?
We have an open issue for more than 1024 files in a folder

Additionally, can  you tell me if there are files with special characters like coma, tilda, and such?
A screen shot of the folder directory (from your laptop) would be great insight

Thanks
Here is our version with (a few) games

[image]

shockdesign, 2019-09-14:
@32teeth There are only 264 roms in the SMS directory and 319 in the GG directory, a few too many to take a screenshot :) My NES directory has more with 785 and GB with 622.

As far as special characters, yes, there are commas, brackets (both open and close) and apostrophe's. However all those characters appear in the directories that work and those that don't.

I've attached the directory listing from the SMS directory.
[sms.txt](https://github.com/retro-esp32/RetroESP32/files/3612721/sms.txt)

Let me know if that helps.

shockdesign, 2019-09-17:
@32teeth Out of curiosity, are there any crash logs written to the SDCard? (Also the crash looks like a reboot of the front end (same as when you exit a game)).

32teeth, 2019-09-17:
if you are familiar with the ptyhon / esp32 sdk, you can take a look at the monitor using ```make monitor``` or alternatively you can simply open arduino and open the serial port 

i have been looking at it here it is most likely either
1. Some unaccounted for special character
2. Unable to sort the files

But most likely #1 above

Thanks for the text dump you sent, there are already a few i know i've missed
* Long Dash
* Special case of a dot in the file name like ```Spy vs. Spy```

Will keep you posted

shockdesign, 2019-09-17:
@32teeth Awesome, I can happily remove those special characters if need be!

32teeth, 2019-09-17:
Testing at home.
I added the whole 539 Sega - Master System Mark III library i have at home

Odroid Go
[image]
[image]

Retro ESP32
[image]

Game List
[sms.txt](https://github.com/retro-esp32/RetroESP32/files/3623458/sms.txt)

Can you confirm if you are running the [latest firmware](https://github.com/retro-esp32/RetroESP32/releases/tag/v.1.0.6-Sound-Fix)?

shockdesign, 2019-09-18:
Can confirm I'm running the latest, have md5 sum'ed the firmware from what is stored on my SDCard compared to the latest (downloaded) and they match `cfb7a8200eae9e03e736afc0e111c77d`

shockdesign, 2019-09-18:
@32teeth Okay a bit more information. I deleted all the ROMS, added them back and it seemed to work at least on SMS, GameGear kind of works, but switching out then switching back to it seems to reboot it, then it works again..

32teeth, 2019-09-18:
@shockdesign can you dump a txt of your ```GG``` folder

```ls > gg.txt```

Do you mean switching as in running the game then back, or just back and forth between sms and gg?

shockdesign, 2019-09-18:
Just switching back and forth between sms and gg..

[gg.txt](https://github.com/retro-esp32/RetroESP32/files/3627007/gg.txt)

32teeth, 2019-09-20:
https://github.com/retro-esp32/RetroESP32/releases/tag/v.1.0.8-Folders

shockdesign, 2019-09-20:
@32teeth Thanks, that didn't help though :( Also noticed PC Engine is odd for me as well.. As in I can scroll through the list of ROMs, but clicking on one it won't show the run button, it will then reboot whilst partially showing the button..

I do wonder if it could be an issue with my memory card?

It's a Sandisk Ultra Class 10 card? Potentially too fast?

32teeth, 2019-09-20:
> PC Engine is odd for me as well.. As in I can scroll through the list of ROMs, but clicking on one it won't show the run button, it will then reboot whilst partially showing the button..

I'll take a look tonight.
Thanks for the update

shockdesign, 2019-09-20:
> > PC Engine is odd for me as well.. As in I can scroll through the list of ROMs, but clicking on one it won't show the run button, it will then reboot whilst partially showing the button..
> 
> I'll take a look tonight.
> Thanks for the update

Also I've plugged the device in to Arduino to check the serial logger..

```
)⸮N⸮!⸮⸮!⸮⸮⸮⸮
SBS9⸮!⸮⸮⸮9⸮=o⸮⸮⸮o⸮
⸮N⸮BBB⸮⸮⸮⸮⸮⸮⸮@cA⸮J⸮⸮O⸮֢fBm⸮ǀ⸮⸮oG⸮ 
⸮!⸮⸮⸮)⸮k允⸮vR⸮z⸮#kR⸮j⸮bR⸮C⸮BǼCR!B⸮cB⸮⸮BD⸮ȅ⸮BB1SC⸮RB@⸮⸮B⸮)"⸮掄RB"⸮⸮!CBa ⸮RG⸮BR⸮BBH ⸮b⸮⸮bB⸮1C⸮bB⸮Cc⸮⸮g⸮c⸮⸮i⸮⸮⸮⸮B⸮⸮b⸮⸮⸮Rb1C⸮s⸮J⸮Rb7C⸮b̷B⸮C⸮k⸮J⸮⸮⸮i⸮R⸮1⸮DbL⸮9
S@⸮⸮v⸮`k⸮f"⸮⸮D⸮⸮b⸮⸮"k⸮s⸮ɭՇ⸮⸮i⸮5-⸮555⸮⸮|⸮⸮⸮⸮Ĝ%j%J⸮⸮O⸮⸮DHc⸮v⸮@⸮⸮b⸮9C⸮RD⸮⸮3⸮允J⸮OB⸮ǆ⸮B)1⸮⸮⸮RB"⸮⸮!⸮BBDHBRBR⸮fBE⸮BBDs⸮DCBB1⸮m!BBB⸮⸮
⸮⸮RB⸮B⸮Ba ̆⸮⸮RB⸮J^⸮bB9⸮S⸮⸮⸮bBD⸮R^RbB⸮Cc⸮⸮⸮Rb3CLCb⸮⸮⸮B⸮⸮b⸮⸮⸮
⸮⸮⸮B⸮J⸮Rb7CSb⸮⸮Rj!⸮⸮⸮⸮⸮⸮O⸮⸮
⸮Χ⸮⸮
⸮c⸮/!⸮⸮1⸮9⸮⸮⸮
⸮⸮!!)
⸮⸮⸮⸮
Ō
9⸮*⸮H0Հ⸮H⸮⸮⸮Ɩ⸮⸮⸮
ż)W⸮⸮⸮⸮⸮!@⸮!v⸮⸮M⸮⸮⸮!⸮⸮⸮⸮)p⸮⸮!⸮1⸮⸮⸮⸮)9⸮⸮⸮⸮⸮P!⸮݌⸮⸮j"⸮Ƭc⸮R⸮iݧ7⸮⸮⸮1
<ň!*⸮!⸮⸮0⸮
⸮⸮⸮⸮!⸮!E⸮
```

@32teeth  it's slightly unreadable tho :( I guess I need to work out what character set it's trying to send data?

32teeth, 2019-09-21:
Run the arduino Serial Monitor @ 115200

shockdesign, 2019-09-21:
@32teeth: that seems to work a bit better... in regards to the GameGear issue..

```
path:/sd/roms/sms
***ERROR*** A stack overflow in task launcher has been detected.
abort() was called at PC 0x4008c764 on core 1

Backtrace: 0x4008c524:0x3ffe7ed0 0x4008c74d:0x3ffe7ef0 0x4008c764:0x3ffe7f10 0x40089c30:0x3ffe7f30 0x4008b4f0:0x3ffe7f50 0x4008b4a6:0x3ffe7f70 0x400d639a:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57
```

And for the PC Engine issue
```
path:/sd/roms/pceGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400ec1af  PS      : 0x00060230  A0      : 0x800d6792  A1      : 0x3ffe81d0  
A2      : 0x00000000  A3      : 0x3ffdc4e4  A4      : 0x3ffdd6b0  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d7d4c  A9      : 0x3ffe81b0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffd37ac  A13     : 0x00000003  
A14     : 0x00000013  A15     : 0x3ffdc4e4  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40090748  LEND    : 0x40090776  LCOUNT  : 0x00000000  

Backtrace: 0x400ec1af:0x3ffe81d0 0x400d678f:0x3ffe81f0 0x400d67a6:0x3ffe8410 0x400d6a3c:0x3ffe8430 0x400d7677:0x3ffe8450 0x400887f1:0x3ffe8570

Rebooting...
ets Jun  8 2016 00:22:57
```

All with the latest 1.0.8 firmware!

32teeth, 2019-09-22:
That helps alot more. 
Do you have anything in the PCE folder? or is it empty?

shockdesign, 2019-09-22:
@32teeth it's not empty, I've attached the file listing below.

Mind you I can look at the ROMS on the device for the PCE  emu, I just cannot run them..

[pce.txt](https://github.com/retro-esp32/RetroESP32/files/3639779/pce.txt)

shockdesign, 2019-09-22:
Let me know if there is anything else I can do to help debug this!

32teeth, 2019-09-23:
The only thing i can think of that i am doing differently is that i am flashing the device directly instead of using firmware.

You can do this as well, but be aware that it will write over the Odroid Firmware (You can always put it back later)

If you are familiar with the espressif framework and have it all up and running
> https://docs.espressif.com/projects/esp-idf/en/latest/get-started/

The you can run the ```ota``` script in the software branch of this repo
> https://github.com/retro-esp32/RetroESP32/tree/Software

Follow the instructions on installation and give it a go.
This is what i run on my odroid go

shockdesign, 2019-09-24:
@32teeth 

This line is currently failing
`git submodule foreach git pull origin master`

due to
```
Entering 'Emulators/odroid-go-pcengine-huexpress'
fatal: couldn't find remote ref master
fatal: run_command returned non-zero status for Emulators/odroid-go-pcengine-huexpress
```

32teeth, 2019-09-24:
@shockdesign 

dang, @pelle7 may have changed the branch name
run ```./branches```

> Emulators/odroid-go-pcengine-huexpress
Should be on ```develop``` [branch](https://github.com/pelle7/odroid-go-pcengine-huexpress/tree/develop)

then run ```./ota```

ill see if i can adjust on my end

shockdesign, 2019-09-24:
@32teeth Out of curiosity, is there a better medium to chat about this process? i.e a slack or discord channel? :)

shockdesign, 2019-09-24:
@32teeth running `git submodule update --recursive --remote` instead ... will try again now..

Last time it wiped the firmware clean and didn't build anything properly, only the bootloader.bin

shockdesign, 2019-09-24:
@32teeth Okay looks like the builds are failing.. hmm

```
                 from /Users/rajesh/src/RetroESP32/Tools/esp-idf/components/cxx/cxx_exception_stubs.cpp:2:
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:466:12: error: '_vfscanf_r' was not declared in this scope
 int _EXFUN(_vfscanf_r, (struct _reent *, FILE *__restrict, const char *__restrict, __VALIST)
            ^~~~~~~~~~
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:466:12: note: suggested alternative: '_mstats_r'
 int _EXFUN(_vfscanf_r, (struct _reent *, FILE *__restrict, const char *__restrict, __VALIST)
            ^~~~~~~~~~
            _mstats_r
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:466:25: error: expected primary-expression before 'struct'
 int _EXFUN(_vfscanf_r, (struct _reent *, FILE *__restrict, const char *__restrict, __VALIST)
                         ^~~~~~
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:466:25: error: expected ')' before 'struct'
 int _EXFUN(_vfscanf_r, (struct _reent *, FILE *__restrict, const char *__restrict, __VALIST)
                        ~^~~~~~
                         )
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:468:12: error: conflicting declaration 'int _EXFUN'
 int _EXFUN(_viprintf_r, (struct _reent *, const char *, __VALIST)
            ^~~~~~~~~~~
In file included from /Users/rajesh/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:18,
                 from /Users/rajesh/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdlib:75,
                 from /Users/rajesh/src/RetroESP32/Tools/esp-idf/components/cxx/cxx_exception_stubs.cpp:1:
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/sys/reent.h:779:19: note: previous declaration as '_reent* _EXFUN'
   struct _reent * _EXFUN(__getreent, (void));
                   ^~~~~~
In file included from /Users/rajesh/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdio:42,
                 from /Users/rajesh/src/RetroESP32/Tools/esp-idf/components/cxx/cxx_exception_stubs.cpp:2:
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:468:12: error: '_viprintf_r' was not declared in this scope
 int _EXFUN(_viprintf_r, (struct _reent *, const char *, __VALIST)
            ^~~~~~~~~~~
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:468:12: note: suggested alternative: '__eprintf'
 int _EXFUN(_viprintf_r, (struct _reent *, const char *, __VALIST)
            ^~~~~~~~~~~
            __eprintf
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:468:26: error: expected primary-expression before 'struct'
 int _EXFUN(_viprintf_r, (struct _reent *, const char *, __VALIST)
                          ^~~~~~
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:468:26: error: expected ')' before 'struct'
 int _EXFUN(_viprintf_r, (struct _reent *, const char *, __VALIST)
                         ~^~~~~~
                          )
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:470:12: error: conflicting declaration 'int _EXFUN'
 int _EXFUN(_viscanf_r, (struct _reent *, const char *, __VALIST)
            ^~~~~~~~~~
In file included from /Users/rajesh/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/sys-include/stdlib.h:18,
                 from /Users/rajesh/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdlib:75,
                 from /Users/rajesh/src/RetroESP32/Tools/esp-idf/components/cxx/cxx_exception_stubs.cpp:1:
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/sys/reent.h:779:19: note: previous declaration as '_reent* _EXFUN'
   struct _reent * _EXFUN(__getreent, (void));
                   ^~~~~~
In file included from /Users/rajesh/.espressif/tools/xtensa-esp32-elf/esp32-2019r1-8.2.0/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdio:42,
                 from /Users/rajesh/src/RetroESP32/Tools/esp-idf/components/cxx/cxx_exception_stubs.cpp:2:
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:470:12: error: '_viscanf_r' was not declared in this scope
 int _EXFUN(_viscanf_r, (struct _reent *, const char *, __VALIST)
            ^~~~~~~~~~
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:470:12: note: suggested alternative: '_mstats_r'
 int _EXFUN(_viscanf_r, (struct _reent *, const char *, __VALIST)
            ^~~~~~~~~~
            _mstats_r
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:470:25: error: expected primary-expression before 'struct'
 int _EXFUN(_viscanf_r, (struct _reent *, const char *, __VALIST)
                         ^~~~~~
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:470:25: error: expected ')' before 'struct'
 int _EXFUN(_viscanf_r, (struct _reent *, const char *, __VALIST)
                        ~^~~~~~
                         )
/Users/rajesh/src/RetroESP32/Tools/esp-idf/components/newlib/include/stdio.h:472:12: error: conflicting declaration 'int _EXFUN'
[... cut ...] y-go $FIRMWARE_PATH/Bins/gnuboy-go.bin 0 19 1441792 smsplusgx-go $FIRMWARE_PATH/Bins/smsplusgx-go.bin 0 20 1703936 stella-go $FIRMWARE_PATH/Bins/stella-go.bin 0 21 786432 prosystem-go $FIRMWARE_PATH/Bins/prosystem-go.bin 0 22 1507328 handy-go $FIRMWARE_PATH/Bins/handy-go.bin 0 23 720896 pcengine-go $FIRMWARE_PATH/Bins/pcengine-go.bin
--------------------------------------------------------------------------
🦄  GOODBYE
--------------------------------------------------------------------------
```

shockdesign, 2019-09-24:
@32teeth if you can build the binaries for an ODROID, and dump a zip of it, should be able to get the ./ota script to work!

32teeth, 2019-09-24:
Unzip the attached
Put them in 
```
[GitHub Repo Folder]/Firmware/Bins/
```

Navigate to 
```
[GitHub Repo Folder]/Scripts/
```

Run 
```./ota```

Answer **N** or **NO** To all questions

It will run this (which will take about 2 minutes to upload)
```
esptool.py \
--chip esp32 \
--port ${PORT} \
--baud ${BAUD} \
--before default_reset \
--after hard_reset write_flash -z \
--flash_mode dio \
--flash_freq 80m \
--flash_size detect \
0x1000 $FIRMWARE_PATH/Bins/bootloader.bin \
0x8000 $FIRMWARE_PATH/Bins/partitions.bin \
0x10000 $FIRMWARE_PATH/Bins/${BIN}.bin \
0x200000 $FIRMWARE_PATH/Bins/${BIN}.bin \
0x280000 $FIRMWARE_PATH/Bins/nesemu-go.bin \
0x340000 $FIRMWARE_PATH/Bins/gnuboy-go.bin \
0x3F0000 $FIRMWARE_PATH/Bins/smsplusgx-go.bin \
0x550000 $FIRMWARE_PATH/Bins/stella-go.bin \
0x6F0000 $FIRMWARE_PATH/Bins/prosystem-go.bin \
0x7B0000 $FIRMWARE_PATH/Bins/handy-go.bin \
0x890000 $FIRMWARE_PATH/Bins/pcengine-go.bin
```
[Bins.zip](https://github.com/retro-esp32/RetroESP32/files/3647982/Bins.zip)

shockdesign, 2019-09-24:
@32teeth Okay have done that... still exhibiting the same issues...

```
***ERROR*** A stack overflow in task launcher has been detected.
abort() was called at PC 0x4008c764 on core 1

Backtrace: 0x4008c524:0x3ffe6500 0x4008c74d:0x3ffe6520 0x4008c764:0x3ffe6540 0x40089bfa:0x3ffe6560 0x4008b4f0:0x3ffe6580 0x4008b4a6:0x00000000

Rebooting...
ets Jun  8 2016 00:22:57
```
(When navigating to the GameGear..)

```
path:/sd/roms/pceGuru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x400ec1af  PS      : 0x00060230  A0      : 0x800d6792  A1      : 0x3ffe81d0  
A2      : 0x00000000  A3      : 0x3ffdc53c  A4      : 0x3ffdd708  A5      : 0x0000ff00  
A6      : 0x00ff0000  A7      : 0xff000000  A8      : 0x800d7d4c  A9      : 0x3ffe81b0  
A10     : 0x00000000  A11     : 0x00000000  A12     : 0x3ffd37ac  A13     : 0x00000003  
A14     : 0x00000013  A15     : 0x3ffdc53c  SAR     : 0x0000000a  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000000  LBEG    : 0x40090748  LEND    : 0x40090776  LCOUNT  : 0x00000000  

Backtrace: 0x400ec1af:0x3ffe81d0 0x400d678f:0x3ffe81f0 0x400d67a6:0x3ffe8410 0x400d6a3c:0x3ffe8430 0x400d7677:0x3ffe8450 0x400887f1:0x3ffe8570

Rebooting...
ets Jun  8 2016 00:22:57
```
(When clicking on a PCE Rom)

So sorry this is a PITA.. I hope it's not something dodgy with the h/w on this particular ODROID Go or something.... 

Saying that, flashing it this way did actually flash properly this time.. so that was a start..

Going to try again by erasing the flash, and flashing at a slower speed.

32teeth, 2019-09-24:
@shockdesign 
No worries, others may have this as well

The Odroid Go site, and Other Crash's recommendation are min 16GB U1 Sd Card. I am using 32GB HC U1, will try older one

mattkj, 2019-10-05:
I'm having the same issue, as mentioned on Reddit.

[gg.txt](https://github.com/retro-esp32/RetroESP32/files/3693472/gg.txt)
[sms.txt](https://github.com/retro-esp32/RetroESP32/files/3693473/sms.txt)

32teeth, 2019-10-07:
@mattkj 
Cleaned to be list only (removed dates and file size)
[gg.txt](https://github.com/retro-esp32/RetroESP32/files/3697675/gg.txt)
[sms.txt](https://github.com/retro-esp32/RetroESP32/files/3697676/sms.txt)

32teeth, 2019-10-07:
@jinlucky @gabozepocaio 

Using
```
#!/bin/bash
EXT=gg
LIST=~/Downloads/${EXT}.txt
echo $LIST

cat $LIST | while read FILE; do
  touch "${FILE%.*}.${EXT}"
done
```

I have created folders (attached zip files) with empty files with the appropriate extensions to test @mattkj and @shockdesign issue

Please test
[gg.zip](https://github.com/retro-esp32/RetroESP32/files/3697878/gg.zip)
[Uploading sms.zip…]()

32teeth, 2019-10-07:
@mattkj 
did you flash this using firmware or via bash script ```./ota```?

Am unable to reproduce using ```./ota``` will attempt to flash as firmware

mattkj, 2019-10-07:
@32teeth 

firmware

halmeeks, 2019-10-28:
I'm having the same problem, but it is with NES and PCE directories. I deleted all but 5 games in my NES directory, and it now works (there were 278 games in the directory). I left my PCE directory alone (352 games), and it still crashes. Based on this thread, I am thinking it is a character in a file name that is causing this. Any suggestions where to start?

32teeth, 2019-10-29:
Solution and download in #30 

[image]
[image]
[image]
[image]

closed 2019-10-29 by 32teeth