retro-esp32/RetroESP32 #25: LM4875 is NO DAC! Use PCM5102a!

ripper121, 2019-09-19:
With the LM4875 you will only amplify the crappy sound from the build in DAC. Will sound like shit.

Please use a real DAC like the PCM5102a, with this you dont even need a preamp. Its powerful enough.

sdorre, 2019-09-19:
Well this guy is building a Retro console. The music coming from the Games is still 8Bits music. 
I would say there is no "real" need for this quality of DAC unless you want to build a real Music player.

On top of that, there are costs (2.86$ against less than 0.80$ for the LM) and complexity to integrate.. 

So, if you want to have especially this DAC on the board, why don't you integrate by yourself (Prototyping + Schematic + Routing + Software + Debug + ...) ?

ripper121, 2019-09-20:
Did you heard the sound on the Odroid Go via Headphones?! Its like crappy tin can.
The ~3$ is worth it. Then you can even use the retro-esp32 as a MP3 Player or a synthesizer.
The LM4875 amplifiys the crappy sound even more :D, so would use it.
I would add it, then you get more interest in it.

There is no Sch or Brd eagle file in the GIT so I cant modify it by my own.

pebri86, 2019-09-20:
Yes, have a multipurpose device is a nice. I use UDA1334A, even cheaper less than $2 worth and great outputs.

ripper121, 2019-09-20:
Sure we can use every other DAC but the PCM5102a is already supported by mostly all Emulators

32teeth, 2019-09-20:
Hi there @ripper121 

Appreciate your insight and feedback. 
If you are comparing the LM4875 from Raspberry Pi to ESP32 there is a significant difference.

I can put a video together for you side by side for comparison and post on Instagram.

As per your request for the SCH and BRD files, this is something we release after production. After which, we welcome individuals such as yourself to build on our projects.

Thank you for your ongoing support.

closed 2019-09-20 by 32teeth

ripper121, 2019-09-20:
ESP32 has only a 8Bit DAC its not made to generate Sound.
For good sound you need 32 Bit Dac

pebri86, 2019-10-02:
Hi @ripper121, we worked together to develop esp32 based game console with integrated DAC.
you can follow the project [Here](https://hackaday.io/project/166707-esplay-micro). I also ported your Odroid Go's MP3 Player DAC version there.

ripper121, 2019-10-02:
Yeah saw that but its not in the Gameboy Pocket housing. (that was my first gameboy and I like the Designe of it the most)