retro-esp32/RetroESP32 #34: Odroid Go Firmware - Rotated Launcher

cjanner, 2019-10-29:
**Describe the bug**
Launcher is rotated 90° to the left in Odroid Go version. Also games are rotated and their image is scrambled.

**To Reproduce**
Running firmware version 20181001 and retro-esp32 version v.1.3.
Odroid-Go Rev. 0.1 20180518

**Screenshot**
[image]

**Additional context**
Seems similar to https://github.com/retro-esp32/RetroESP32/issues/16.

32teeth, 2019-10-29:
This was entirely my fault. Compiling to different versions at the same time.

Please find i have updated the [Odroid Go](https://github.com/retro-esp32/RetroESP32/releases/download/v.1.3-PC-Engine-Frameskip/odroid-go.zip) firmware.

closed 2019-10-29 by 32teeth

cjanner, 2019-10-29:
Yes I can confirm. Now it is working.
Thanks