retro-esp32/RetroESP32 #35: External DAC support on Odroid Go

Ayolland, 2019-11-05:
The Go Play firmware supports external DACs, allowing the addition of headphone jack without hardware hacking. This option is toggled o/off in Go Play by holding the volume button and pressing the right button while on the main menu, and indicated by the icon next to the volume bars.

AFAIK, this is not supported currently by the Retro ESP32 firmware, which is is unfortunate if you want to use the ODroid in any kind of audio setup. (LSDJ, nanoloop, etc.) It would nice to see if this was supported in some way.

32teeth, 2019-11-05:
Thanks @Ayolland 
Do you have an example of the hardware for the expansion port?
IIRC the onboard DAC chip for the expansion makes a difference.

LSDJ, Nanoloop...  ...cool, never see Odroid Go implementations of this. Will research more.

We will leave this Issue unassigned for now, and mark it as an enhancement.

Ayolland, 2019-11-05:
Sure. 
I think the most common one is the PCM5100A DAC, since it is used in [the only hat AFAIK that has been designed specifically for the Odroid Go](https://backofficeshow.com/shop/odroidgohat).

I also have seen [a video](https://www.youtube.com/watch?v=cKPM9KuZo8E&feature=youtu.be) where someone hooked up a PCM5102A DAC using [this board.](https://www.amazon.com/gp/product/B075B44CL6/)

ripper121, 2019-11-22:
+1 add the PCM5100A Dac to the HW and FW support.

atylmo, 2019-12-13:
Ditto. Would love to see it added. I have that Backoffice hat.

32teeth, 2020-04-01:
Closing to icebox

closed 2020-04-01 by 32teeth

h1aji, 2022-04-01:
Hi @32teeth any chance to implement this feature?

h1aji, 2022-04-01:
Described here https://forum.odroid.com/viewtopic.php?t=31853

Have tested ODROID Go with UDA1334a DAC, works great