retro-esp32/RetroESP32 #37: Games won't show up for zx-spectrum and c64

lortschi, 2019-11-14:
Games won't show up for zx-spectrum and c64.

**To Reproduce**
Steps to reproduce the behavior:
1. Newest firmware cloned down and installed all the dependencies.
2. Rom folders spectrum and c64 are available. Drivers are included.
3. Games are included to the folders.
4. Retro-ESP32 is flashed and started.
5. No games in zx-spectrum menu folder. C64 even no menu folder.

**Expected behavior**
Emulators menu folders should be available on demand. Games should be loaded properly into the folders.

32teeth, 2019-11-14:
thanks @lortschi 

Currently C64 is not implemented as a selectable Emulator (working on onscreen keyboard) although there is a C64 folder

Spectrum games in ```spectrum``` folder should have ```.zx``` extension

lortschi, 2019-11-17:
thanks @32teeth for the quick answer. But actually I tried with the .zx file extension to the spectrum games and still no success. The spectrum game is already .tzx file type. So I tried it to rename to .zx or to extend it with .tzx.zx

Or did I understand you wrong maybe?

32teeth, 2019-11-19:
Can you give the latest [Firmware](https://github.com/retro-esp32/RetroESP32/releases/tag/v.1.5-Brightness-Control) a shot?
Thanks

lortschi, 2019-11-19:
Brightness, Color and Volume control fine and awesome! But still Zx-spectrum games aren't recognized. One thing I forget to mention, in the Zx-Spectrum menu if is start pressed, it's starting to load... something and freezes. After this froze restarted OS won't boot up, just blank screen and need to be flashed again.

32teeth, 2019-11-19:
If you get a blank screen trying to load spectrum game, you can still press menu to bring up the in game menu system

32teeth, 2019-11-20:
https://github.com/retro-esp32/RetroESP32/releases/tag/v.1.6-ZX-Spectrum-Fix

closed 2019-11-20 by 32teeth

lortschi, 2019-11-20:
Sorry for bothering you guys, but this bug is quite not solved. I tested the last release and it's only shows .z80 Sinclair games. That is not ZX-Spectrum ;) The ZX-Spectrum 48k Homecomputer has .tzx file type. These files are actually not shown up in the menu and of course not playable.

lortschi, 2019-11-20:
Sorry I was wrong and it's working fine. The .z80 files are now visible and playable. Tried just .tzx files coz I thought for the zx-spectrum 48k are these the common files. But there are also as .z80 file type available. All good and thanks for this fast fix! Thumbs up for this awesome odroid flash version. I like a lot your work.

Pitel, 2019-11-30:
How about .tap?