retro-esp32/RetroESP32 #38: When scrolling through emulators, "NO GAMES AVAILABLE" shows up randomly for everything and do not reappear until system reboot.

evan64, 2019-11-21:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
Scroll right on main menu.

**Expected behavior**
Games should show up.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: v.1.6

32teeth, 2019-11-21:
Hi @evan64 
thanks for the Issue Log

This was happening previously, can you let me know which version of the firmware you are using?

[https://github.com/retro-esp32/RetroESP32/releases/latest](Can you try this on the latest)

evan64, 2019-11-21:
It's happening on 1.6, the latest.

32teeth, 2019-12-03:
@evan64 we patched the latest release
if this is still happening, please post me a video or image of the sequence to reproduce

metapy123, 2019-12-11:
I can reproduce this error. Here are the steps I took and they result in the error 100% of the time.
I have over 2500 games between systems and when I start going between consoles the games disappear when I get to the 2500 or so game. You can also have 800 games on one system go to the system then back to settings and repeat a few times for the error to appear.

So to replicate have Retro ESP count 2500+ games.

I noticed I went to NES games after settings after first boot. 864 NES games, back to settings, back to NES 864 games, back to settings, back to NES 779 games, then no games at all. I then deleted some games to get under 2500 total and then I could navigate to all the systems, but as soon as I hit 2500+ all games disappear even when the total games were under 2500.

I hope this helps and happy to do further testing as needed.

32teeth, 2019-12-17:
I am able to reproduce this.
Am setting this issue as higher priority.

Thanks

32teeth, 2019-12-23:
Fixed in [v.1.7](https://github.com/retro-esp32/RetroESP32/releases/tag/v.1.7-Memory-Heap-Fix)

closed 2019-12-23 by 32teeth