retro-esp32/RetroESP32 #39: Schematics

tomvdb, 2019-11-22:
Hi There,

I love this project, I have a few odroid go's, but this is perfect since I can put this into my broken gameboy shell. I had some boards made from your gerbers and busy collecting the parts. I can't seem to find a schematic or the eagle design files anywhere? Maybe I'm looking past them? I found the eagle lbr and the gerber files, but nothing else. Is the schematic and/or eagle files available ?

Thanks,
Tom

ripper121, 2019-11-22:
Im also interested in the Eagle SCH and BRD files. I like to add a DAC.

32teeth, 2019-11-22:
@tomvdb 
Please find i have added a [PDF](https://github.com/retro-esp32/RetroESP32/blob/Hardware/eagle/RetroESP32.pdf) of the schematics.

@ripper121 
The [PDF](https://github.com/retro-esp32/RetroESP32/blob/Hardware/eagle/RetroESP32.pdf) of the schematics is available.

Here's the deal. We keep the SCH/BRD files private until first production run. After that, we will release the files. This was the same case as for our other projects

tomvdb, 2019-11-22:
Thanks for the pdf schematic. It's all the text is mirrored though :p

When are you guys planning first production run? Would be keen to order an assembled one

32teeth, 2019-12-02:
@tomvdb 
Currently in production
Delivery scheduled for End Of Year
Market January 2020

closed 2019-12-02 by 32teeth

livigni, 2021-01-26:
Hi People,
any chances the Eagle sch and brd files to be released?
Also, the aliexpress link for the display is dead, could you indicate an alternative?
Thanks.