retro-esp32/RetroESP32 #43: NES Emulation Issues (Classic Dragon Quest 4 issue)

derpman4k, 2019-12-13:
I don't have much to say in terms of technical reasons why, but on the Odroid Go certain NES games (in this case Dragon Quest 4) won't even boot. I'm not sure if this is an issue with the emulator core being used, or if a core can be changed to get around this, or if it's just a limitation of the hardware, or some optimization issues, or what have you (Being into emulators for a long time, I just know some games are always a pain to emulate). 

But a friend of mine has always wanted to run this on his Odroid Go and I figured I'd ask. Also curious if this will be an issue on the RetroESP32 hardware as well. 

Thanks!

v666, 2019-12-15:
Same happens with Kick Master.

32teeth, 2020-02-25:
Tested in  [v.2.1](https://github.com/retro-esp32/RetroESP32/releases/tag/v.2.1-NES%2BLYNX)

closed 2020-02-25 by 32teeth

mrjoshw408, 2020-02-26:
Tested on 2.1 with US version of games:
Dragon Warrior 3
Dragon Warrior 4

I still get the grey screen and does not continue further.