retro-esp32/RetroESP32 #44: UI crashes when loading PC Engine ROM

jdalmanza, 2019-12-13:
**Describe the bug**
When launching PCE / TG16 ROM UI goes dark and restarts after a few seconds.

i'm not sure if this is an actual bug, or a documentation issue, but I do not see any documentation for the PCE emulator. Perhaps I'm doing something wrong? 

**To Reproduce**
roms are stored in folder named PCE and display as expected in the UI menu
Browsing PCE/TG16 roms, select game to run, screen dims and UI resets

**Expected behavior**
Do not get emulator menu (restart, resume, delete, etc) as I do with other emulators before crash

**Additional context**
ROMs are in .pce format, not compressed.

Wend1go, 2019-12-13:
Does it happen with all ROMs?
On my side, I most PC Engine ROMs play fine (just saving doesn't work)  but some of them cause the screen to turn gray and even after switching the device off and on again the gray screen persists.
In that case I need to reflash the complete FW.

jdalmanza, 2019-12-13:
It happened for all roms I tried.

Using no-intro romset verified sometime in 2018. I tested with Bonk 1 ~3,
alien crush, air zonk, blazing lasers, sidearms, and ninja spirit (my
favorites, btw). All attempts resulted in the same behavior.

The same ROMs work just fine on my retropie setup, so I have verified that
they are not corrupt. The only difference is that they are zipped in
retropie, unzipped here

atylmo, 2019-12-13:
This is happening to me too. The standalone firmware emulator works, but no games work inside Retro ESP32. Granted, I've only tried two :P

I thought maybe I needed a BIOS like I do for the Lynx, but after some looking around I believe you only need one for CD games.

vistamix77, 2019-12-15:
I am having the same problem, with both gameboy and PCE directories. It tries the load the directory, dims and then restarts. In addition, the lynx directory is showing no files. When loading the lynx directory, it crashed bad enough that I had to reflash the firmware. I'm currently using go-emu and it's working fine.

theVanZan, 2019-12-16:
Think the folder in "odroid/data" has to be there for pce and have the same name as the folder in roms that contains your games.

atylmo, 2019-12-16:
@theVanZan Good call! I forgot about doing that. I bet I didn’t need the Lynx BIOS either. That fixed my problem! Thank you :)

That would explain why the other ones worked. I already had those folders from using other firmware.

theVanZan, 2019-12-16:
Yeah same here. The wiki really should have all this kind of info. I spent a while trying to get Spectrum games to show up until I learned that they have to be z80 files.

32teeth, 2019-12-16:
Thank you everyone for you insight
We are working on a wiki

It is great to see conversations here in the community

theVanZan, 2019-12-16:
Thank you for Retro ESP32....I really like it. Great work.

vistamix77, 2019-12-16:
I didn't have /data/pce so created it, and it didn't make a difference. So, I then decided to try a different microSD card. What I was using was a 32 GB Sandisk Ultra HC 1. I then tried a Kingston 4GB HC card. With this card, initially all the directories except Lynx show content, but when I cycle back through them again, all the directories are empty. Once I power off and back on they continue to show up as empty. I reformatted the card using SDFormatter and tried again, same thing. I then reflashed the firmware with go-emu, and everything worked. Weird. I am tempted to go buy another microSD card. What are people using?

32teeth, 2019-12-17:
Our apologies to those experiencing the reboot issue.

It seems that since we had been working with our own code base for so long, it slipped our mind that we had been working with a specific [Odroid Go Firmware Version (esp-idf-v3.2 branch)](https://github.com/OtherCrashOverride/odroid-go-firmware/releases/tag/20180718).

## Preperation
> Update Odroid Go Firmware

[Follow the guide here](https://medium.com/@siloraptor/upgrading-the-firmware-of-the-odroid-go-dd7eebdc7ed4) 

Again, our apologies.

jdalmanza, 2019-12-17:
I did not have the directories for the "new" emulators in the odroid/data folder on the memory card. Once I added those, it works just fine. ... slow, but fine.

closed 2019-12-17 by jdalmanza