retro-esp32/RetroESP32 #46: Lynx emulator has poor sound compared to the firmware version of it

atylmo, 2019-12-13:
**Describe the bug**
The Lynx emulator that is included with Retro ESP32 has sound glitching that isn't present in the standalone firmware version of it.

**To Reproduce**
Steps to reproduce the behavior:
1. Open a game with the Lynx emulator (I'm using California Games).
2. Listen to the music.
3. Flash the standalone firmware (https://github.com/pelle7/odroid-go-handy/releases).
4. Load the same game and compare the sound. It's much better in the standalone release. Is this because Retro ESP32 is using an older build of it?

**Expected behavior**
The sound from the standalone emulator and the one included in Retro ESP32 should match.

32teeth, 2020-01-21:
resolved in [Lynx Sound Quality](https://github.com/retro-esp32/RetroESP32/releases/tag/v.1.9-Lynx-Sound-Quality-Fix) release

using @pelle7 release

closed 2020-01-21 by 32teeth

atylmo, 2020-01-22:
This is fixed (thank you :D) but every time I run a Lynx game the system shuts itself off. Once I toggle the power switch and it boots back into the Lynx emulator, the game works fine. However, when I exit the emulator, it will happen again the next time I load a Lynx game.