retro-esp32/RetroESP32 #47: System reebot when selecting Sega Master System

EduardoOrtegaME, 2019-12-14:
**Describe the bug**
I don't have any game for the Sega Game Gear System and when I'm trying to select this system the Odroid Reboots.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Sega Game Gear
2. See error

**Screenshots**
Check the video with the problem here:

https://drive.google.com/open?id=1uSRk8C_VuYwg9vrqcMAM9YWtMr5Mz7p2

v666, 2019-12-15:
I think that happens with all emulators that don't have any roms available. Same happened to me with PC Engine.

EduardoOrtegaME, 2019-12-15:
I have other systems with no roms at all as you can see in the video and works perfectly... Gonna add dome roms to see if that's the problem...

shockdesign, 2019-12-16:
This sounds familiar to the issue I was having @32teeth 

@EduardoOrtegaME What version of the firmware are you running?

EduardoOrtegaME, 2019-12-19:
I’m using Version 1 Build 6 (Version 1.6)

32teeth, 2019-12-23:
Fixed in [v.1.7](https://github.com/retro-esp32/RetroESP32/releases/tag/v.1.7-Memory-Heap-Fix)

commit 9e88f4e7 (2019-12-19): Thu 19 Dec 2019 16:13:33 EST - fixed #47 for next release
closed 2019-12-23 by 32teeth