retro-esp32/RetroESP32 #52: Lynx black screen

DrBuns3n, 2020-02-03:
Tried Batman Returns and Stun Runner in the Lynx emulator (RetroESP32 1.9) and both start to black screen. Switch off Odroid Go and back on again and the game displays ok, but screen brightness is at minimum. Menu brightness change option is not available.

EduardoOrtegaME, 2020-02-06:
Same problem here, I just created a ticket:

https://github.com/retro-esp32/RetroESP32/issues/54

ultrasteveg, 2020-02-23:
Confirmed to have exactly the same issue when using v2.0

32teeth, 2020-02-25:
Resolved in [v.2.1](https://github.com/retro-esp32/RetroESP32/releases/tag/v.2.1-NES%2BLYNX)

closed 2020-02-25 by 32teeth

ultrasteveg, 2020-02-25:
Not resolved for me in v2.1 - have posted a video link in comment on other ticket.

EduardoOrtegaME, 2020-02-25:
Me neither, screen still turning off when selecting an Atari Lynx game.

EduardoOrtegaME, 2020-02-28:
Solved in v2.2