retro-esp32/RetroESP32 #54: Screen turns off when selecting Atari Lynx game

EduardoOrtegaME, 2020-02-06:
**Describe the bug**
Once I select an Atari Lynx game the screen turns off.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to 'Atari Lynx'
2. Select a game
3. See error

**Expected behavior**
The game launches correctly.

**Screenshots**
You can see a video here: https://drive.google.com/open?id=121NM-n6VXreRm_kneZ295M7Dw4pT_3GG

**Additional context**
I have version 1.9 installed.
77 games on Atari Lynx emulator.
More than 300 games in the SD.
Using a Odroid GO.
I tried with GameBoy, NES, and GameBoy Color games and this emulators are working OK.

32teeth, 2020-02-25:
Resolved in [v.2.1](https://github.com/retro-esp32/RetroESP32/releases/tag/v.2.1-NES%2BLYNX)

closed 2020-02-25 by 32teeth

ultrasteveg, 2020-02-25:
Issue is still present for me running v2.1

Video here: https://drive.google.com/open?id=1jszKU40C81zTbIfQPjMhjgy95ZYeQok7

EduardoOrtegaME, 2020-02-25:
Not solved, still present in v2.1

EduardoOrtegaME, 2020-02-28:
Solved in v2.2