retro-esp32/RetroESP32 #56: where to get the bootloader for the retro esp32 board ? i flashed my board with wrong bootloader .. and now the screen is the wrong way ;(

lastclass74, 2020-02-20:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

32teeth, 2020-02-20:
The are located in the [releases](https://github.com/retro-esp32/RetroESP32/releases) section

[zip file](https://github.com/retro-esp32/RetroESP32/releases/download/v.2.0-Cover-Art-Option/retro-esp.zip)

[direct download](https://github.com/retro-esp32/RetroESP32/blob/Software/Firmware/Releases/retro-esp/Retro%20ESP32.fw?raw=true)

Now, to restore it

1. add the above ```.fw``` file onto the sd card
2. insert sd card
3. power on _while_ holding **B**
4. you will now be back in the core firmware
5. navigate to the new firmware and flash it

as a tip
restarting while holding B will always get you back to the core firmware

closed 2020-02-20 by 32teeth

lastclass74, 2020-02-21:
thx but its the base firmware for retro esp32.. can i just you odroid base firmware  ?

thx for the help

32teeth, 2020-02-21:
Not sure i understand

The Retro ESP32 comes with [Retro Odroid Go Firmware](https://github.com/retro-esp32/Retro-Odroid-Go-Firmware) Installed

Retro ESP32 is a firmware you load on top

If you are looking to use the standard Odroid Go Firmware from Crash, you can find it in the Software branch of the repo

32teeth, 2020-02-21:
Oh, NVM, you have an Odroid Go?
Yes, the release page always has both versions, just download the right one
 
## Odroid Go **THIS IS WHAT YOU HAVE**
> 2.4" Display

Launcher to the **Odroid Go** for the ILI9341 LCD Driver
#### [odroid-go.zip](https://github.com/retro-esp32/RetroESP32/releases/download/v.2.0-Cover-Art-Option/odroid-go.zip)

## Retro ESP32
> 2.6" Display

Launcher to the **Retro ESP32** *(Game Boy Pocket)* ILI9342 LCD Drive
#### [retro-esp.zip](https://github.com/retro-esp32/RetroESP32/releases/download/v.2.0-Cover-Art-Option/retro-esp.zip)

lastclass74, 2020-02-21:
no im sorry if im hard to understand..
i need Retro Odroid Go Firmware (.img) for the retro esp32 - 2.6" Display ... 

thx

32teeth, 2020-02-25:
If you where able to ```flash``` your device using USB

[Go here](https://github.com/retro-esp32/Retro-Odroid-Go-Firmware)

1. clone repo
2. cd into ```Firmware``` folder
3. Run ```make menuconfig```
4. change your settings appropriately
5. go to ```Scripts``` folder
6. run ```./ota```

lastclass74, 2020-02-25:
thats the one i looked for ... now works 100 % thx ...

hope to see c64 soon ;)