retro-esp32/RetroESP32 #59: Odroid don't start when adding more than 50+ games in Fav

EduardoOrtegaME, 2020-02-28:
**Describe the bug**
Adding more than 50 games between various systems (GB, GBC, Lynx, etc.) as favorites make the system collapse, and you turn the Odroid on, the creen blinks and won't start.

**To Reproduce**
Steps to reproduce the behavior:
1. Add more than 50 games as favorites between various systems.
2. Turn off, turn on.
3. See error

**Screenshots**
Se the problem here:
https://drive.google.com/open?id=1kmSb3esOcrFU1r19Sz5f8Qrqv0MoPxFZ

And see the list of favorites here:
https://drive.google.com/open?id=1O1t0ZEkmNgozieETxDyL-dOvj-BGeL2t

**Odroid go (please complete the following information):**
 - Version v2.2

**Additional context**
Once I opened the esp-32-txt file and deleted some of the games form the list, the system booted up correctly.

32teeth, 2020-02-28:
@EduardoOrtegaME 

Thanks for the heads up.
This was by design

```
    free(FAVORITES);
    FAVORITES = (char**)malloc((50) * sizeof(void*));
```

We can change this.
What would you consider a tangible number?

or

Should we add a safeguard to limit to 50 favorites?

32teeth, 2020-02-28:
To Do

- [ ] Increase Limit of Favorites?
- [ ] Safeguard if over 50
- [ ] ???

EduardoOrtegaME, 2020-02-28:
Oh didn't know that... if this is an intended behavior maybe once we reach this limit the add to favorite feature shouldn't do anything

obsidianspider, 2020-03-07:
If this is by design (performance limitations, etc.) then I think that when the limit is reached the user should be informed or as suggested, that no new favorites can be added. This would be more elegant than the system failing to boot.

closed 2020-03-16 by 32teeth