retro-esp32/RetroESP32 #62: add c64 emulator to retro esp32 ;)

lastclass74, 2020-03-02:
**Is your feature request related to a problem? Please describe.**
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

**Describe the solution you'd like**
A clear and concise description of what you want to happen.

**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

lastclass74, 2020-03-19:
no ?

bastetfurry, 2021-01-02:
With luck someone could take Frodo, it worked very good back then on my nGage GD which is far less powerful than the ESP32, but don't expect 100% compatibility. Many demos and some games using demo effects might break apart.