retro-esp32/RetroESP32 #65: Volume button behavior should be more consistent.

kindfulkirby, 2020-03-11:
**Is your feature request related to a problem? Please describe.**
Some emulators (NES, PCE, Lynx) have a menu that opens when pressing the Volume button (:speaker: symbol, to the right of the Menu button), which is great!
Other emulators (GB, GBC) just turn the sound on and off, which is also acceptable, and you can change the volume by holding the Volume button and pressing up or down.
But the other emulators (except for maybe ZX Spectrum, which I can't get to work) *increase* the volume when pressing the Volume button, and "circle back" to silence only after having gone to **maximum volume**, which is very annoying when you're trying to turn the volume *down*.

**Describe the solution you'd like**
Making the volume button in the "other" emulators work like a simple mute button (i.e. just turn audio on or off) would be a good start (you can always save and change the volume in the main settings menu). Making it work like on the GB(C) emulators would be even better, and having a nice on-screen menu for all emulators would of course be perfect, but probably hard to implement.

**Describe alternatives you've considered**
Doing it the "old way" technically works, but the Odroid Go can get awfully loud, so having to pass by maximum volume in order to lower the volume can be a bit scary, especially late at night!

**Additional context**
If I have the time, I might look into this myself, but I'm not very optimistic about that right now.

32teeth, 2020-03-11:
This i great idea.
We are looking to add the volume slider to all the HUD for each emulator

As a patch in the interim, the next release has a fix to all the levels in all the emulators being the same values

32teeth, 2020-06-26:
Volume Control in HUD available in latest release (2.5 as of writting)

closed 2020-06-26 by 32teeth