retro-esp32/RetroESP32 #68: looks like sega (gamegear and mastersystem) dont work in v2.3 but works in v2.2 ?!

lastclass74, 2020-03-19:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

32teeth, 2020-03-19:
Hi

In order to triage the issue better, please fill out the form appropriately

thanks

lastclass74, 2020-03-19:
Describe the bug : sega emulator dont work in version 2.3 (works on 2.2)
Expected behavior : i like it to work again ;)
Screenshots : white screen when starting a sega rom
os : RETRO ESP32
Device: RETRO ESP32
Version: [2.3]

32teeth, 2020-03-19:
@lastclass74 

Are you building this from scratch or is this the monitor from the provided .fw file?

If you are building from scratch
Official firmware release uses **smsplusgx-go** from **retro-go** folder in *emulators*

Before compiling locally you must
1. Run the ```./replace``` script
2. Run the ```./ota``` script and select new version ```retro-esp``` NOT ```odroid-go```

lastclass74, 2020-03-19:
i downloaded it from : https://github.com/retro-esp32/RetroESP32/releases/download/v.2.3-Recently-Played/retro-esp.zip
is this the monitor from the provided .fw file? Yes

lastclass74, 2020-03-21:
works now in version 2.4 :) great thanks ..

closed 2020-03-21 by lastclass74