retro-esp32/RetroESP32 #69: Gameboy games battery saves

shar33f, 2020-03-24:
**Describe the bug**
Some Gameboy games feature in game saves that allow to save game progress and resume at a later time. I am not able to resume Gameboy games using this built in native game save (battery) option. I.e. in Mario's Picross pressing start during gameplay and choosing the "Save" option does not save the games position. I believe this is because RetroESP32 does not currently support Gameboy battery saves

**To Reproduce**
Steps to reproduce the behavior:
1. Launch Gameboy "Mario's Picross" (https://en.wikipedia.org/wiki/Mario%27s_Picross)
2. During gameplay press Start button and then choose the in game Save option
3. The game displays a message "Your game has been saved. You can continue later."
4. Close the ROM and restart the odroid-go or alternately restart the game by pressing Menu button then choosing RESTART GAME
5. The game will start over but NOT give the option to resume at the point it was last saved (from Step 3 above)

**Expected behavior**
On restart, the game should prompt "Get ready to continue your saved game." but there is no prompt; the save point is lost and you are forced to start the game from the beginning again. Other Gameboy emulators like Gambatte will create a .sram file and allow the game to continue where it was last saved.

**Screenshots**
Press start to see the Save screen:
[image]

The is the screen shown after choosing Save:
[image]

This is the expected screen to allow resume at last saved position. I am able to get to this screen if I DO NOT restart the odroid-go.
[image]

**Additional context**
I am running the latest RetroESP32 v2.4.
Link to Libretro Gambatte emulator information: https://docs.libretro.com/library/gambatte/
Link to Gambatte github: https://github.com/libretro/gambatte-libretro

32teeth, 2020-03-24:
Thanks for the in depth  response
We have been looking into restoring save files that the actual game produces in battery

For the time being, you will need o use the in game menu

* Menu Key on Odroid Go (What you have in the pictures)
* Start+Select on Game Boy Retro ESP32

Select Save Game / Overwrite Game

Again, thank you, please know we continue to work on this

shar33f, 2020-03-24:
You're welcome, and BTW my level of detail is directly proportionate to the level of boredom I have being stuck at home 😆. First of all, thank you so much for the hard work you are putting into the firmware for this amazing little device! You guys are doing a great job!! Now, for me, ideally I would have access to the raw .srm file that is written when I choose to save my game, this way I can easily copy the .srm between devices / emulators and have the ability to resume my game on any of my devices. Are you planning on writing out the raw .srm file to disk in future?

32teeth, 2020-03-26:
LOL, no kidding with WFH.

The way most of the emulators work is they produce a ```.sav``` file not an ```.srm``` the way RetroPie / Lakka / RetroArch do

It's something worth looking into.

shar33f, 2020-03-27:
Yes, I know exactly what you mean. I think those .sav in other emulators are .state. The thing is, I have found that sometimes these .sav and .state are not always compatible between different emulators, whereas the .srm always is. So for me, just makes things easier and simpler to be able to copy the .srm file between any system or emulator and not have to worry about converting it. Just my use case, and I guess this isn't really an issue unless you have lots and lots of different devices and emulators 🙄 lol

PGuitarnut, 2020-04-24:
Wow love this thread! Yes please consider making the sav/srm files transferrable between platforms! I would buy this product in a heartbeat if this was possible as the specs on your Odroid Advance are very nice!