retro-esp32/RetroESP32 #70: A couple of ideas

DisableGraphics, 2020-03-30:
**GBA emulation**
No, i'm not crazy. One month ago I found an emulator (mGBA, to be more concise) that worked on the ESP32 chip. The only drawback is that it isn't fast. In fact, it is VERY slow. The source code is here: https://github.com/ganhailin/mgba-esp32. It isn't mine, and, since this guy deserved an award, why not pay homage to him in the very best odroid go firmware?
**Background image**
Of course, with some limitations (max resolution: 320 * 240) since the ESP32-WROVER-B chip lacks memory.
**Remembering configurations**
Since the favourites and recents are stored in a text file, why not also remember the configurations?
**Save & Exit command**
I have deleted thousands of times the progress of a game clicking in the incorrect selection. It would be perfect.

32teeth, 2020-03-30:
There seem to be a few items here.

## GBA
> There was a youtube posted in January, 

[7FPS on YouTube](https://www.youtube.com/watch?v=2g8WL2x0BaU)

## Background Image
> More Detail Needed

You can definitely do a background image, it would have to be processed in chunks to the frame buffer

Can you describe further what you want?

Like Lakka Background / Screen Saver?

## Remembering Configurations
> Which Specifically?

Which conifg specifically?
Volume, Brightness, Icon Colorization, Theme are all written to NVS Storage

## Save & Exit
> A better solution

Is it the vocabulary?
What would you do?

DisableGraphics, 2020-03-30:
Yes, like Lakka background.

 When i reflash the firmware, all the brightness, color, icon colorization configuration resets. That kind of configuration is what i'm talking about.

Refering to the Save & Exit, there is actually a save button, and an exit button. I'm referring to a combination of those two, it saves and then exits.

I'm not a native english speaker, so I might make errors. Thank you.

32teeth, 2020-03-31:
## Save & Exit
> A better solution

Combination Option in List

* [x] Resume Game
* [x] Restart Game
* [x] Save Game
* [ ] Save & Exit Game <-- NEW OPTION
* [ ] ~~Exit~~ **Quit** Game

DisableGraphics, 2020-04-04:
Yes, exactly.

closed 2023-02-28 by DisableGraphics