retro-esp32/RetroESP32 #77: lsdj.gb does not work

OdroidGuy, 2020-06-06:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

32teeth, 2020-06-07:
Hi @OdroidGuy and welcome to the project.
Our team is not familiar with **_lsdj.gb_**

Would you be able to provide appropriate information as to 'does not work' as in title

Please see our issue form for intake. Thank you

-----
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.
-----

32teeth, 2020-06-10:
Closing due to no response from @OdroidGuy

closed 2020-06-10 by 32teeth

ucodelukas, 2020-07-24:
LSDJ is a native tracker for the Gameboy
i will test it on Monday