retro-esp32/RetroESP32 #82: Roms fail to load after reaching 50 items on recently played list

klundry, 2020-08-04:
Roms fail to load once you go over 50 games in your recents list with the caveat that it WILL load if it is already in your recents list. I believe it is related to #59 

There is a clear recents option but would it be possible to just drop the oldest item off the list to prevent this?

32teeth, 2020-08-11:
@klundry thanks, we will look into this for the next release.
Much Appreciated insight

32teeth, 2020-08-31:
Closed as of latest release

closed 2020-08-31 by 32teeth

klundry, 2021-01-04:
I'm on the latest release but ran into this issue again with the recents list.

EugeneAndruszczenko, 2021-01-05:
We are aware of this. And working on it.
For the time being you can easily just go to the 'recents' and 'clear recents' and your good for another 50 games :(
Sorry i know it's a pain in the ass, but we will have a fix up soon

klundry, 2021-01-05:
OK, cool. Yeah, its not too big a deal just thought I would mention it since the issue was closed.

commit 8a816e8a (2021-01-05): Tue Jan  5 12:51:05 EST 2021 - prep for 2.8 fix #82
commit a3930883 (2021-01-05): Tue Jan  5 12:54:02 EST 2021 - prep for 2.8 fix #82