retro-esp32/RetroESP32 #83: make menuconfig is missing

cimttja, 2020-08-05:
**Describe the bug**
Hi, I cloned your repository to start some work on the issue ["Only Show Emulators With Files"](https://github.com/retro-esp32/RetroESP32/issues/23). With the instructions in your [DIY.md](https://github.com/retro-esp32/RetroESP32/blob/master/DIY.md),  I made it to setup my enviroment, do first changes in the code, compile, build a new image and use it on my Odroid-Go. 
But when I try to setup the menue as described in [make menuconfig](https://github.com/retro-esp32/RetroESP32/blob/master/DIY.md#retro-esp32) it fails.

**To Reproduce**
Steps to reproduce the behavior:
```
cd Launchers/retro-esp32
make menuconfig
```
this results in 
```
install@install-VirtualBox:~/RetroESP32/Launchers/retro-esp32$ make menueconfig 
WARNING: esp-idf git submodule components/esp32/lib may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/mbedtls/mbedtls may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/esptool_py/esptool may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
Toolchain path: /home/install/esp/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
Toolchain version: crosstool-ng-1.22.0-80-g6c4433aCompiler version: 5.2.0
GENCONFIG
WARNING: esp-idf git submodule components/esp32/lib may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/mbedtls/mbedtls may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
WARNING: esp-idf git submodule components/esptool_py/esptool may be out of date. Run 'git submodule update' in IDF_PATH dir to update.
make: *** Keine Regel, um „menueconfig“ zu erstellen.  Schluss.
(translated: there's no rule to build „menueconfig“)
```

**Expected behavior**
I would expect the [described menue screen](https://github.com/retro-esp32/RetroESP32/blob/master/DIY.md#retro-esp32) to open and configure the odroid-go button behavior

**Desktop:**
 - OS: Ubuntu
 - Version 18.04.04 LTS

**Additional context**
It seems, that the $IDF_PATH is set correctly.
```
install@install-VirtualBox:~/RetroESP32/Launchers/retro-esp32$ echo $IDF_PATH
/home/install/RetroESP32/Tools/esp-idf
```
where "install" is the user on the machine.

32teeth, 2020-08-11:
Thanks @cimttja 
From the log (thank you for sharing) posted, seems to be an xtensa compiler (gcc) issue not esp-idf
Check this [link](https://docs.espressif.com/projects/esp-idf/en/release-v3.0/get-started/linux-setup.html#toolchain-setup) for setting up the toolchain

sdorre, 2020-08-11:
I think that is just a typo 
you entered `make menueconfig`
What happen if you try `make menuconfig`

cimttja, 2020-08-14:
Hi, thanks for your help. 

The typo in my previous description was not causing my issue but i reinstalled the esp-idf as described in your link and now i can access  the ```make menuconfig``` but I don't get the option to choose the menu behavior.

[image]

I'm a bit confused, but isn't the installer in the Scripts directory doing the same things as the manual installation? As I can see, even the binaries seems to be the same.

32teeth, 2020-08-17:
The core launcher only has those two options.

It's in the ```kconfig``` file in the ```components``` folder

```
menu "Retro ESP32 Configuration"
    choice LCD_DRIVER_CHIP
        prompt "LCD Driver Chip"
        default LCD_DRIVER_CHIP_ODROID_GO
        help
            LCD Screen Driver

    config LCD_DRIVER_CHIP_ODROID_GO
        bool "Odroid Go - 2.4\" [ILI9341]"

    config LCD_DRIVER_CHIP_RETRO_ESP32
        bool "Retro ESP32 - 2.6\" [ILI9342]"

    endchoice

    choice MENU_HOT_KEYS
        prompt "MENU Key Options"
        default DEFAULT_MENU_KEY
        help
            Select Hot Key (Menu Button)

    config DEFAULT_MENU_KEY
        bool "Odroid Go MENU Button"

    config COMBO_MENU_KEY
        bool "START + SELECT as MENU button"

    endchoice
endmenu    
```

closed 2020-09-16 by 32teeth