retro-esp32/RetroESP32 #84: Blank screen after flashing - system doesn't work

Artur2048, 2020-10-11:
Right after installing RetroESP32 on my Odroid GO the screen goes blank and system doesn't respond to any buttons. Restarting device and trying various button combinations (like holding menu while powering on) didn't helped.

I have the latest Odroid-GO firmware: **Ver 20181001**:
https://github.com/OtherCrashOverride/odroid-go-firmware/releases/tag/20181001

I'm using the latest RetroESP32 **release V2.6**:
https://github.com/retro-esp32/RetroESP32/releases/download/v.2.6-Recent-Favorite-Fix/odroid-go.zip

Steps to reproduce the behavior:
1. download odroid-go.zip (v2.6 of RetroESP)
2. put "Retro ESP32.fw" in odroid/firmware folder on the SD card
3. hold B button while powering on Odroid GO
4. choose ESP firmware, click A, then START to confirm

Verification, Erasing and flashing starts - in total 8 cycles of erase/write.
At the end screen goes blank.

(device works perfectly fine on the original "GO-Play v. 20181004")

32teeth, 2020-10-11:
Hmm, I just tested on mine and was unable to reproduce

Shut off the device
Hold B
Turn device on still holding B
It will restore to Odroid Firmware

Artur2048, 2020-10-11:
Hello,
Thank you for your response,
Yes. I know about "B" - power combination.
It of course works and let's me reflash my device again and again. No
problem.

I have tried few times to flash RetroESP32 without success.

I've made a video:
https://www.dropbox.com/s/chc63w7sroqttfb/20201011_202513_1_1.mp4?dl=0

Best​ ​regards,
-Artur

lsignoret, 2020-10-14:
Hi! 

Just FYI I got mi Odroid Go yesterday in the mail and I'm having the exact same issue as you. If you can resolve the issue please let us know Arthur. Thanks!

Artur2048, 2020-10-15:
Hello,

Unfortunately I wasn't able to get it working.

Let's compare our hardware, maybe it's related to particular board version.

I have:
"10th anniversary, rev 0.1 20180807".

Best Regards,
-Artur

smutek110, 2020-12-12:
This bug its only on boards witch 8Mb PSRAM

bodobear, 2020-12-13:
Hi,
I just tried to install the actual version and it´s also not working ony my new odroid go.
got the same revision like Artur mentioned before. "10th anniversary, rev 0.1 20180807"
Using the multi bootloader and also retro-Go. everything is working fine but not RetroESP32.
Have another old Retroid Go Model.. no problems on that one so it has to be the newer/different board.
Any plans or infos about a fix for that issue ?
Thank you... Bodo

bropple, 2020-12-21:
I'm also having this problem. Latest firmware with a REV 0.1 20180807 board. After attempting to flash the firmware using the B button boot menu the system's screen just goes black and stays that way until I reflash the original go-play firmware.

I thought it could have something to do with the SD card skeleton files, but mine are the same as the one in the repo.

bropple, 2020-12-21:
Redoing this post with some more information:
It seems the firmware has an issue with the SPI RAM on the "REV 0.1 20180807" ODROID-GO. It continuously tries to reboot itself after running into this error, resulting in a blank screen:

> Rebooting...
ets Jun  8 2016 00:22:57
>
>rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:6724
load:0x40078000,len:8900
ho 0 tail 12 room 4
load:0x40080000,len:7388
entry 0x4008039c
I (30) boot: ESP-IDF v3.1-odroid-1-g08f2c8f26 2nd stage bootloader
I (31) boot: compile time 09:24:57
I (31) boot: Enabling RNG early entropy source...
I (37) qio_mode: Enabling default flash chip QIO
I (42) boot: SPI Speed      : 80MHz
I (46) boot: SPI Mode       : QIO
I (50) boot: SPI Flash Size : 16MB
I (54) boot: Partition Table:
I (58) boot: ## Label            Usage          Type ST Offset   Length
I (65) boot:  0 nvs              WiFi data        01 02 00009000 00004000
I (73) boot:  1 otadata          OTA data         01 00 0000d000 00002000
I (80) boot:  2 phy_init         RF data          01 01 0000f000 00001000
I (88) boot:  3 firmware         factory app      00 00 00010000 000f0000
I (95) boot:  4 retro-esp32      OTA app          00 10 00100000 00080000
I (103) boot:  5 nesemu-go        OTA app          00 11 00180000 000c0000
I (110) boot:  6 gnuboy-go        OTA app          00 12 00240000 000b0000
I (118) boot:  7 smsplusgx-go     OTA app          00 13 002f0000 00160000
I (125) boot:  8 spectrum         OTA app          00 14 00450000 00080000
I (133) boot:  9 stella-go        OTA app          00 15 004d0000 001a0000
I (140) boot: 10 prosystem-go     OTA app          00 16 00670000 000c0000
I (148) boot: 11 handy-go         OTA app          00 17 00730000 00170000
I (156) boot: 12 pcengine-go      OTA app          00 18 008a0000 000b0000
I (163) boot: End of partition table
I (167) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x1a270 (107120) map
I (205) esp_image: segment 1: paddr=0x0011a298 vaddr=0x3ffb0000 size=0x05d78 ( 23928) load
I (212) esp_image: segment 2: paddr=0x00120018 vaddr=0x400d0018 size=0x29d68 (171368) map
I (257) esp_image: segment 3: paddr=0x00149d88 vaddr=0x3ffb5d78 size=0x0b208 ( 45576) load
I (271) esp_image: segment 4: paddr=0x00154f98 vaddr=0x40080000 size=0x00400 (  1024) load
I (272) esp_image: segment 5: paddr=0x001553a0 vaddr=0x40080400 size=0x0b1b0 ( 45488) load
I (304) boot: Loaded app from partition at offset 0x100000
I (304) boot: Disabling RNG early entropy source...
I (305) spiram: Found 4095MBit SPI RAM device
I (309) spiram: SPI RAM mode: flash 80m sram 80m
I (315) spiram: PSRAM initialized, cache is in low/high (2-core) mode.
I (322) cpu_start: Pro cpu up.
I (326) cpu_start: Starting app cpu, entry point is 0x40081288
I (317) cpu_start: App cpu up.
I (336) heap_init: Initializing. RAM available for dynamic allocation:
I (343) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM
I (349) heap_init: At 3FFD9858 len 000067A8 (25 KiB): DRAM
I (355) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (362) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (368) heap_init: At 4008B5B0 len 00014A50 (82 KiB): IRAM
I (374) cpu_start: Pro cpu start user code
I (379) spiram: Adding pool of 0K of external SPI memory to heap allocator
E (387) cpu_start: External RAM could not be added to heap!
abort() was called at PC 0x4008124a on core 0
>
>Backtrace: 0x4008952b:0x3ffe3c10 0x40089825:0x3ffe3c30 0x4008124a:0x3ffe3c50 0x4008145b:0x3ffe3c80 0x40078d99:0x3ffe3ca0 0x4007914e:0x3ffe3ce0 0x40080477:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

usepinlove, 2021-01-18:
I have the same problem...
Is there any solution?

Artur2048, 2021-01-19:
I haven't found any solution anywhere.
It seems to be a bug in RetroESP32 that shows up only on certain Odroid-go
boards (probably the most recent ones)
-Artur

stl1988, 2021-02-09:
Same here. Rev 20180807 and black screen with the same externam RAM to heap error.

32teeth, 2021-02-09:
Correct, this is if you are on the old firmware.
There is a great article here https://medium.com/@siloraptor/upgrading-the-firmware-of-the-odroid-go-dd7eebdc7ed4

I am working on the base firmware as well

stl1988, 2021-02-09:
That link is not related to RetroESP32. Go-Play is not affected by that black screen issue.

32teeth, 2024-07-17:
HI @nottyl 

I can recommend you port to @ducalex firmware https://github.com/ducalex/retro-go

closed 2024-07-17 by 32teeth