retro-esp32/RetroESP32 #85: Can you change the LCD driver?Ili9342 is no longer available in China。

jianjinliao, 2020-10-19:
Can you change the LCD driver?Ili9342 is no longer available in China。

fizarum, 2020-10-19:
You can buy it here, for example:
https://a.aliexpress.com/_B1R9Ul

jianjinliao, 2020-10-19:
this is 37pin LCD，SPI interface is not supported

jianjinliao, 2020-10-19:
51pin ili9341 LCD is no longer available in China，I bought an ili9341 37pin LCD and found that I couldn't use it without SPI interface

32teeth, 2020-10-19:
We are working on an ILI9342C version, please be patient with us

jianjinliao, 2020-10-20:
Thank you for your reply ， I'll wait for your new version

jianjinliao, 2020-10-21:
hi,Can PCB files be shared? I want to change the shape, 3D print

fizarum, 2020-10-21:
hey PCB, BOM, etc.  can be found here: https://github.com/retro-esp32/RetroESP32-Design-Files

32teeth, 2020-10-23:
@jianjinliao What are you changing the shape to?

jianjinliao, 2020-10-23:
I didn't find the BRD file, I want to change to rectangle similar, I don't have a GB shell

32teeth, 2020-10-23:
> I didn't find the BRD file, I want to change to rectangle similar, I don't have a GB shell

That is correct, the BRD file is currently not available, this is due to changing to the newer 45 PIN ILI9342C

That being said, we have not left you alone. Sounds like you are only looking for the Gameboy Pocket (MGB-001) PCB dimensions.

You can find the library file [here](https://github.com/retro-esp32/RetroESP32-Design-Files/blob/master/lbr/RetroESP32.lbr#L2689)

[image]

32teeth, 2020-10-23:
Closing Issue as link to library and assets provisioned.

closed 2020-10-23 by 32teeth