retro-esp32/RetroESP32 #86: uCity crashes the emulator

Pitel, 2020-10-20:
**Describe the bug**
uCity crashes the emulator right after start

**To Reproduce**
Steps to reproduce the behavior:
1. [Download uCity](https://github.com/AntonioND/ucity/releases/tag/v1.2)
2. Start it
4. The emulator crashes

**Expected behavior**
The game runs

**Additional context**
It used to work in previous versions.

Pitel, 2020-10-21:
- 2.3 bad
- 2.2 good

m-a-r-c-h, 2020-10-26:
I confirm this bug concerning several roms on my Odroid Go with firmware 2.6. After the crash the micro-sd isn't recognized anymore and the red icon is displayed while booting. It's the same icon when turning the devide on when no card is inserted. Holding down the menu button while booting brings the system back to life so I can boot another firmware.

AntonioND, 2023-02-16:
I think it's because of the SRAM size of the game. It's valid, but used only by one or two games if I remember correctly. Can you try the `ucity_compat.gbc` ROM from here? https://github.com/AntonioND/ucity/releases/tag/v1.2 That one has been reduced to a more common size.