retro-esp32/RetroESP32 #89: Manual

n1trux, 2020-12-06:
Would be nice if you could write a manual on how to use this.

I wanted to gift this to a kid and had no time to explain :(

32teeth, 2020-12-06:
Thank you for the feedback.

Unfortunately We currently do not have any stock on these items and thus want be able to provide you any packaging or printed material.

If you are looking for additional information and have fully reviewed the provided direction in our Wikipedia section, please reference the original Odroid Go documentation for more technical details

We appreciate your patronage

n1trux, 2020-12-06:
well, at least a small manual on how to use this "for dummies" would be cool for the README.md. This has nothing to do with the original Odroid hardware, because it's the RetroESP32 menu where the documentation is missing.

E.g.:

> ← / → : Change tabs (console, settings select)
> ↑ / ↓ : select game
> A : open game / open game menu
> B : back
> ≡: menu / back to main menu
> 🔈: change ingame settings (Volume etc.)
> SELECT / START: scroll through menu pages

I don't want a package or a printed manual, I just wanted to gift my old ODROID GO with your firmware to a 9yo kid ;)

For anyone reading this: the ODROID GO can still be bought from the German vendor Pollin.de: https://www.pollin.de/p/odroid-go-mobile-spielekonsole-kit-810887

Is there a way to create screenshots from the system to enrich a manual with them? E.g. "this is the main menu, you can do X and Y here".

32teeth, 2021-01-05:
https://github.com/retro-esp32/RetroESP32/blob/master/USAGE.md

closed 2021-01-05 by 32teeth