retro-esp32/RetroESP32 #90: Problem witch 8Mb PSRAM Boards

smutek110, 2020-12-11:
Is there any fix for motherboards with 8Mb PSRAM? Because there is PSRAM initialization error and system reboot on motherboards with 8Mb PSRAM.

32teeth, 2020-12-12:
@smutek110 
We are going to move this issue to 'To Be Considered' state

Both the Odroid Go and Retro ESP32 have support for external PSRAM and size of 16MB

Recently working with the CuteESP32 team (latest firmware) we have made it available for less PSRAM and internal support

To do this manually, simply enter into the Menu Config of each emulator and change the ESP32 Component Specific feature for the memory

You can run the ```./ota``` script provided to reconfigure

smutek110, 2020-12-12:
I used t8_v1.7.1 with 16MB-flash/8MB-PSRAM and i get error initialize psram in retroESP32.

20:56:30.005 -> ets Jun  8 2016 00:22:57
20:56:30.005 -> 
20:56:30.005 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
20:56:30.005 -> configsip: 0, SPIWP:0xee
20:56:30.005 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
20:56:30.005 -> mode:DIO, clock div:1
20:56:30.005 -> load:0x3fff0018,len:4
20:56:30.005 -> load:0x3fff001c,len:6692
20:56:30.005 -> load:0x40078000,len:6556
20:56:30.005 -> load:0x40080000,len:9616
20:56:30.005 -> entry 0x40080420
20:56:30.039 -> [0;32mI (29) boot: ESP-IDF v3.1-beta1-43-g49fbef7 2nd stage bootloader[0m
20:56:30.039 -> [0;32mI (29) boot: compile time 00:10:55[0m
20:56:30.039 -> [0;32mI (38) boot: Enabling RNG early entropy source...[0m
20:56:30.039 -> [0;32mI (38) qio_mode: Enabling QIO for flash chip WinBond[0m
20:56:30.039 -> [0;32mI (40) boot: SPI Speed      : 80MHz[0m
20:56:30.039 -> [0;32mI (45) boot: SPI Mode       : QIO[0m
20:56:30.039 -> [0;32mI (49) boot: SPI Flash Size : 16MB[0m
20:56:30.039 -> [0;32mI (53) boot: Partition Table:[0m
20:56:30.039 -> [0;32mI (56) boot: ## Label            Usage          Type ST Offset   Length[0m
20:56:30.073 -> [0;32mI (64) boot:  0 nvs              WiFi data        01 02 00009000 00004000[0m
20:56:30.073 -> [0;32mI (71) boot:  1 otadata          OTA data         01 00 0000d000 00002000[0m
20:56:30.073 -> [0;32mI (78) boot:  2 phy_init         RF data          01 01 0000f000 00001000[0m
20:56:30.073 -> [0;32mI (86) boot:  3 firmware         factory app      00 00 00010000 000f0000[0m
20:56:30.073 -> [0;32mI (93) boot:  4 retro-esp32      OTA app          00 10 00100000 00080000[0m
20:56:30.073 -> [0;32mI (101) boot:  5 nesemu-go        OTA app          00 11 00180000 000c0000[0m
20:56:30.141 -> [0;32mI (108) boot:  6 gnuboy-go        OTA app          00 12 00240000 000b0000[0m
20:56:30.141 -> [0;32mI (116) boot:  7 smsplusgx-go     OTA app          00 13 002f0000 00160000[0m
20:56:30.141 -> [0;32mI (124) boot:  8 spectrum         OTA app          00 14 00450000 00080000[0m
20:56:30.141 -> [0;32mI (131) boot:  9 stella-go        OTA app          00 15 004d0000 001a0000[0m
20:56:30.141 -> [0;32mI (139) boot: 10 prosystem-go     OTA app          00 16 00670000 000c0000[0m
20:56:30.141 -> [0;32mI (146) boot: 11 handy-go         OTA app          00 17 00730000 00170000[0m
20:56:30.141 -> [0;32mI (154) boot: 12 pcengine-go      OTA app          00 18 008a0000 000b0000[0m
20:56:30.141 -> [0;32mI (161) boot: End of partition table[0m
20:56:30.141 -> [0;32mI (166) esp_image: segment 0: paddr=0x00100020 vaddr=0x3f400020 size=0x1a270 (107120) map[0m
20:56:30.175 -> [0;32mI (203) esp_image: segment 1: paddr=0x0011a298 vaddr=0x3ffb0000 size=0x05d78 ( 23928) load[0m
20:56:30.175 -> [0;32mI (210) esp_image: segment 2: paddr=0x00120018 vaddr=0x400d0018 size=0x29d68 (171368) map[0m
20:56:30.242 -> [0;32mI (256) esp_image: segment 3: paddr=0x00149d88 vaddr=0x3ffb5d78 size=0x0b208 ( 45576) load[0m
20:56:30.242 -> [0;32mI (269) esp_image: segment 4: paddr=0x00154f98 vaddr=0x40080000 size=0x00400 (  1024) load[0m
20:56:30.242 -> [0;32mI (270) esp_image: segment 5: paddr=0x001553a0 vaddr=0x40080400 size=0x0b1b0 ( 45488) load[0m
20:56:30.309 -> [0;32mI (302) boot: Loaded app from partition at offset 0x100000[0m
20:56:30.309 -> [0;32mI (302) boot: Disabling RNG early entropy source...[0m
20:56:30.309 -> [0;32mI (303) spiram: Found 4095MBit SPI RAM device[0m
20:56:30.309 -> [0;32mI (308) spiram: SPI RAM mode: flash 80m sram 80m[0m
20:56:30.309 -> [0;32mI (313) spiram: PSRAM initialized, cache is in low/high (2-core) mode.[0m
20:56:30.309 -> [0;32mI (320) cpu_start: Pro cpu up.[0m
20:56:30.309 -> [0;32mI (324) cpu_start: Starting app cpu, entry point is 0x40081288[0m
20:56:30.309 -> [0;32mI (316) cpu_start: App cpu up.[0m
20:56:30.309 -> [0;32mI (334) heap_init: Initializing. RAM available for dynamic allocation:[0m
20:56:30.376 -> [0;32mI (341) heap_init: At 3FFAE6E0 len 00001920 (6 KiB): DRAM[0m
20:56:30.376 -> [0;32mI (347) heap_init: At 3FFD9858 len 000067A8 (25 KiB): DRAM[0m
20:56:30.376 -> [0;32mI (354) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM[0m
20:56:30.376 -> [0;32mI (360) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM[0m
20:56:30.376 -> [0;32mI (366) heap_init: At 4008B5B0 len 00014A50 (82 KiB): IRAM[0m
20:56:30.376 -> [0;32mI (373) cpu_start: Pro cpu start user code[0m
20:56:30.376 -> [0;32mI (377) spiram: Adding pool of 0K of external SPI memory to heap allocator[0m
20:56:30.410 -> [0;31mE (385) cpu_start: External RAM could not be added to heap![0m
20:56:30.410 -> abort() was called at PC 0x4008124a on core 0
20:56:30.410 -> 
20:56:30.410 -> Backtrace: 0x4008952b:0x3ffe3c10 0x40089825:0x3ffe3c30 0x4008124a:0x3ffe3c50 0x4008145b:0x3ffe3c80 0x400787d5:0x3ffe3ca0 0x40078b86:0x3ffe3ce0 0x400804fb:0x3ffe3df0 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20
20:56:30.410 -> 
20:56:30.410 -> Rebooting...
20:56:30.410 -> ets Jun  8 2016 00:22:57
 But in board with 16MB-flash/4MB-PSRAM retroESP32 works fine.

32teeth, 2020-12-13:
Thanks for the monitor output

For your **LILYGO® TTGO**, 
When you run the ```./ota``` script, select ```cute-esp32``` as your version

Additionally, take a look at the [Configs Folder](https://github.com/retro-esp32/RetroESP32/tree/master/Configs)
It contains all the overwrites for your setup

What you will need to do, as mentioned in the [Do It Yourself](https://github.com/retro-esp32/RetroESP32/blob/master/DIY.md) section, is to run the ```./replace``` [script](https://github.com/retro-esp32/RetroESP32/blob/master/DIY.md#replace-files-for-ili9342-drivers)

The CuteESP32 is built on the same board you have (https://www.instagram.com/p/CID-KzmHaI5/)

smutek110, 2020-12-14:
Diy build works fine for mi. Thanks

32teeth, 2020-12-14:
Thats great news @smutek110 
https://github.com/sponsors/retro-esp32

closed 2021-01-03 by 32teeth