retro-esp32/RetroESP32 #93: Eagle files

livigni, 2021-01-26:
Hi People,
any chances the Eagle sch and brd files to be released?
Also, the aliexpress link for the display is dead, could you indicate an alternative?
Thanks.

32teeth, 2021-01-26:
hi @livigni 

### Great question
> This is the Software and Firmware repository

The hardware one is [here](https://github.com/retro-esp32/RetroESP32-Design-Files)
Currently it contains the old version whose LCD is no longer available

We will soon add the IPS ILI9342C which is a 45pin FPC Connector. That version we have removed the FTDI chip to reduce cost as well.

In the meantime...  ...

-----

## Support
> Although we do this for 💕 of 🕹️<br/>
> **We Also Like Coffee!** <br/>
> *Please Consider Supporting* <br/>

Did you know, you can support us in a few way? We offer multiple sposorship tiers, with different rewards!

Take a *look*. 
See what *fits*.

 [image]

**Thank You**

livigni, 2021-01-27:
thanks for the reply and sorry for posting in the wrong place anyway.
The hw repository only has the schematic in pdf version, not the Eagle files form what I see.
Do you currently have an ETA (just a rough one) for the new he revision? Just thinking if building this nice evil device or getting a classic odroid go meanwhile.
thanks.

closed 2021-01-27 by 32teeth