retro-esp32/RetroESP32 #94: Header Match Error

basketballsteve, 2021-04-23:
I have a new ESPlay micro. I have tried to use RetroESP32 FW update to add the additional emulators and new menu interface. I copied the unzipped FW update into the firmware folder of the microsd card but when I boot to the menu with the original firmware, the musicplayer, and then the retroesp32 fw update than i have downloaded and copied onto the microsd card upon selecting it I get the message Header match error. I have tried cloning the original microsd card unto another microsd card but get the same result, i have tried various versions of the fw update all with the same result. I am using a windows 10 computer for coping the downloaded files to the microsd card. 
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Desktop (please complete the following information):**
 - OS: [e.g. iOS]
 - Browser [e.g. chrome, safari]
 - Version [e.g. 22]

**Smartphone (please complete the following information):**
 - Device: [e.g. iPhone6]
 - OS: [e.g. iOS8.1]
 - Browser [e.g. stock browser, safari]
 - Version [e.g. 22]

**Additional context**
Add any other context about the problem here.

32teeth, 2021-05-14:
Do you have somewhere i can take a look at you program structure?

basketballsteve, 2021-05-14:
(851) #New Version# ESPlay Micro V2 - ESP32 Handheld Game Console - YouTube
<https://www.youtube.com/watch?v=d2LnUSSiilY&t=58s>

I don't know how to help you do that. I have the ESPlay V2 and it runs off
the included card that came with it. The only modification that I have done
it with it is to download a few ROMs (colecovision roms copied to the
correct roms folder) and I have downloaded the FW update file and placed it
in the firmware folder.
I took the download from here. Releases · retro-esp32/RetroESP32 · GitHub
<https://github.com/retro-esp32/RetroESP32/releases>
I have tried older and newer versions of this file (same result)
When I load the boot load menu I see

 (game icon) esplay-retro-emu
(music icon) mp3-player_DAC
(no icon) Retro ESP32

Same thing no matter which version of the Retro ESP32 file I use.
I can try to flash that file but it just reports write error and then at
next boot I must reflash the original esplay-retro-emu to be back to square
1.

My interest was to add some other emulators to the unit (Colecovision
Atari) but it does work with the included emulation just doesn't update.

I tried cloning the original sd card onto a new card but that yielded the
same set of errors.