retro-esp32/RetroESP32 #101: How was RetroESP32-Game-And-Watch.bin created from the source?

x130844, 2022-01-01:
for the life of me, I can't find docs on how to create .bin files. only fw files.
Any instructions on creating .bin files fro this source? 
The OTA fails. at the end flashing it, but it does create some files. I do see a lot of red text flashing by, looks like fatal errors. 
I'm missing something but unsure what, I think I followed all the steps on how to compile the src on my own, but it's not working. it just says "Bye" at the end of OTA and nothing is written to the esp32. 
Any documentation you could recommend to get unstuck?  been 3 days now of research and tries, reinstall the toolchain, esp32 tools, trying different repos etc. 
Thank you!

x130844, 2022-01-10:
days later, and after decided to try a docker for esp32 env. fully configured.   they all failed with the instructions shipped with retroEsp.   (using ./OTA, or make, or rg_tool.py or other ).  has anyone ever been able to compile into a .bin from src?

shar33f, 2022-01-10:
Sorry if this is a dumb question but am I supposed to be able to flash RetroESP32-Game-And-Watch.bin file to my odroid-go and then I'll have a game&watch emulator (that would be so bloody sweet!!). I thought I'm only able to flash .fw files..

x130844, 2022-01-10:
> 

as far as I know .fw for retro is if you want to update your firmware from Odroid itself,   you would copy the fw on sdcard and let Odroid update the firmware. 
For the bin file, I was able to flash bin with esptool from computer with USB cable connected to the esp32.  and that worked perfectly. I think the game&watch bin will completely remove the original Odroid retro firmware but I could be wrong here, I'm only using an esp32 hardware. not Odroid GO.

shar33f, 2022-01-10:
I see. I wish there was a .fw version I could flash directly on my odroid-go, maybe one will get released in the future or the game&watch emulator will get merged into a future .fw release 🤞🏻

x130844, 2022-01-10:
I think I was able to compile a .fw for game&watch from src,  but with all my attempts and different IDEs, I'll see if I can recreate it for you. don't hold your breath though.

32teeth, 2022-01-10:
@shar33f 

Grab the [.bin file](https://github.com/retro-esp32/RetroESP32/releases/download/v.2.8-Ampersand-Game%26Watch/RetroESP32-Game-And-Watch.bin) change it to `.fw`

x130844, 2022-01-10:
> There is already a prepared bin for the Game and Watch version.
>
> python2 esptool.py --chip esp32 --baud 2000000 --port [your port] write_flash 0x00 RetroESP32-Game-And-Watch.bin
> If you are building from scratch, i recommend you follow The DIY Instructions <https://github.com/retro-esp32/RetroESP32/blob/master/DIY.md>

Thank you for the reply,   yes and the .bin works fine,  I wanted to start making some changes for custom hardware and others.
Hence why I wanted to recompile.  And I did follow this doc too.   I will try one more time, but I can’t see what I’m missing.

x130844, 2022-01-10:
Hi again,oOk, the steps provided are definitely broken, 

It fails right at the 3rd git command for the submodules. 

# git submodule foreach git pull origin master
Entering 'Emulators/frodo-go'
From github.com:OtherCrashOverride/frodo-go
 * branch            master     -> FETCH_HEAD
Already up to date.
Entering 'Emulators/go-play'
From github.com:OtherCrashOverride/go-play
 * branch            master     -> FETCH_HEAD
Already up to date.
Entering 'Emulators/odroid-go-handy'
From github.com:pelle7/odroid-go-handy
 * branch            master     -> FETCH_HEAD
fatal: refusing to merge unrelated histories
fatal: run_command returned non-zero status for Emulators/odroid-go-handy

I tried adding "--allow-unrelated-histories”
But failed again:
# git submodule foreach git pull origin master --allow-unrelated-histories
Entering 'Emulators/frodo-go'
From github.com:OtherCrashOverride/frodo-go
 * branch            master     -> FETCH_HEAD
Already up to date.
Entering 'Emulators/go-play'
From github.com:OtherCrashOverride/go-play
 * branch            master     -> FETCH_HEAD
Already up to date.
Entering 'Emulators/odroid-go-handy'
From github.com:pelle7/odroid-go-handy
 * branch            master     -> FETCH_HEAD
Already up to date!
Merge made by the 'recursive' strategy.
Entering 'Emulators/odroid-go-pcengine-huexpress'
fatal: couldn't find remote ref master
fatal: run_command returned non-zero status for Emulators/odroid-go-pcengine-huexpress

h1aji, 2022-01-11:
Get your mac and remove all your previous esp-idf and xtensa installations. 
Create ~/esp folder and simply run:
```
cd ~/esp
git clone -b master --single-branch git@github.com:retro-esp32/RetroESP32.git --recursive
cd RetroESP32
git submodule update --init --recursive
git submodule foreach git pull origin master
chmod -R 777 Scripts
cd Scripts
./installer
```

Works fine

x130844, 2022-01-11:
ok, I will try again, the only difference is that I run these commands from my home (~). not ~/esp.  I don't see why it would make a difference, but I'm about to try again. I assume you ran these steps above in the past few days? (just confirming that no changes done on GitHub is what's blocking me. and if the issue is only with me, that's fine, I'll eventually figure it out, I just need to know that indeed it is only me.) 
Thank you for your help.

x130844, 2022-01-11:
Same,  I will try again from a brand new account, I will create. I really don't understand what's going on. why it would work for you but not me. 
=
Unpacking objects: 100% (3/3), 1.37 KiB | 349.00 KiB/s, done.
From github.com:OtherCrashOverride/odroid-go-firmware
 * branch            296aa5b81d96248bcb658adbb5a7c855ab523364 -> FETCH_HEAD
Submodule path 'Odroid/odroid-go-firmware': checked out '296aa5b81d96248bcb658adbb5a7c855ab523364'
eric@erics-MBP esp % ls
RetroESP32
eric@erics-MBP esp % cd RetroESP32 
eric@erics-MBP RetroESP32 % ls
Arduino			Components		Emulators		Launchers		SD Card			Tools.zip
Assets			Configs			Firmware		Odroid			Scripts			USAGE.md
CODE_OF_CONDUCT.md	DIY.md			LICENSE.md		README.md		Tools
eric@erics-MBP RetroESP32 % git submodule update --init --recursive
eric@erics-MBP RetroESP32 % git submodule foreach git pull origin master
Entering 'Emulators/frodo-go'
From github.com:OtherCrashOverride/frodo-go
 * branch            master     -> FETCH_HEAD
Already up to date.
Entering 'Emulators/go-play'
From github.com:OtherCrashOverride/go-play
 * branch            master     -> FETCH_HEAD
Already up to date.
Entering 'Emulators/odroid-go-handy'
remote: Enumerating objects: 2, done.
remote: Total 2 (delta 0), reused 0 (delta 0), pack-reused 2
Unpacking objects: 100% (2/2), 146 bytes | 73.00 KiB/s, done.
From github.com:pelle7/odroid-go-handy
 * branch            master     -> FETCH_HEAD
fatal: refusing to merge unrelated histories
**fatal: run_command returned non-zero status for Emulators/odroid-go-handy**

x130844, 2022-01-11:
Exactly the same error with a brand new account on the laptop.
I guess, next is I'll start a new linux VM.

x130844, 2022-01-11:
ok, linux VM, new github keys, 

root@debian:~/esp/RetroESP32# **uname -a
Linux debian 4.19.0-18-arm64 #1 SMP Debian 4.19.208-1 (2021-09-29) aarch64 GNU/Linux**

apt-get install git  as this is a new environment, 
Then it fails the same way at the end 

root@debian:~/esp/RetroESP32# git submodule update --init --recursive

root@debian:~/esp/RetroESP32# **git submodule foreach git pull origin master**
Entering 'Emulators/frodo-go'
From github.com:OtherCrashOverride/frodo-go
 * branch            master     -> FETCH_HEAD
Already up to date.
Entering 'Emulators/go-play'
From github.com:OtherCrashOverride/go-play
 * branch            master     -> FETCH_HEAD
Already up to date.
Entering 'Emulators/odroid-go-handy'
From github.com:pelle7/odroid-go-handy
 * branch            master     -> FETCH_HEAD
**fatal: refusing to merge unrelated histories
fatal: run_command returned non-zero status for Emulators/odroid-go-handy**
uname
Are you guys sure it's me?

h1aji, 2022-01-11:
Its you. It tells you exactly whats the problem. There is no "master" branch Emulators/odroid-go-handy and in Emulators/odroid-go-pcengine-huexpress. Delete these folders and do git clone
How about deleting your unrelated unformatted comments from https://github.com/retro-esp32/RetroESP32/issues/102 ?

x130844, 2022-01-12:
I removed the comments from 102.
As far as this, I did mention that I could go further by cleaning the folders, but then I get other errors, I fix them, then there are more errors that I have to fix, 
The point being, all my environments make me have to fix about 15 things, I then give up, clearly something is wrong on my side, especially if you guys keep telling me that the guide works 100%. 
If the guide worked as indicated, I shouldn't have to delete these folders and git clone. 
Something is amiss.
If it's me, like you just said, that's perfectly fine. 
the guide is 100% correct, and I need to find why if I follow it to the letter it fails for me while it works for you.
I'm ok with that, h1aji,   Thank you for confirming that the guide works and I'm not following it correctly.

x130844, 2022-01-12:
and for sh*ts and giggles, after removing the folders and trying again, it went further, but then fails with:

fatal: No url found for submodule path 'Launchers/esplay-retro-emulation' in .gitmodules
I can fix that too, but then it's another error. ad-nauseam. I already spent 6 days on this, and I will keep trying for a few more, I will find out what I'm doing wrong.

h1aji, 2022-01-12:
This repo here https://github.com/pelle7/odroid-go-pcengine-huexpress doesnt have master branch, only develop.

You can ask developer why he deleted master branch.

I`d recommend reinstall everything and skip
`git submodule foreach git pull origin master
`
and continue with ./install

x130844, 2022-01-12:
Thank you for confirming that the documentation is indeed incorrect (not up to date) so they would fail. 
Skipping this step allowed to complete, but a lot of fatal errors in red. (going way too fast to actually read, and not sure where the logs/traces are getting saved for me to review) and no new .bin  or .fw generated.
I appreciate the help.

h1aji, 2022-01-12:
I didnt say that documentation is incorrect. I said that mentioned repository doesnt have master branch

x130844, 2022-01-12:
:).  ok, never mind, I can see this is a touchy subject,  Thx again.

x130844, 2022-01-12:
Update:   for anyone else who might be interested in compiling this, the docs are not mentioning that you also need to install ffmpeg as a pre-req. 
I'll faff around a bit more, but I think I'm done with this project.

32teeth, 2022-01-12:
> Update: for anyone else who might be interested in compiling this, the docs are not mentioning that you also need to install ffmpeg as a pre-req. I'll faff around a bit more, but I think I'm done with this project.

**ffmpeg** you can easily install view *brew*
```
brew install ffmpeg 
```

it's only ever used to create the distributable firmware `.fw` files from this repo
this is done by executing the `./firmware` script from the scripts folder

a bit X/Y problem space here
Please follow the issue markdown as provided for future use
it really does help with our intake
[image]
e

x130844, 2022-01-12:
Hi, can you delete this whole GitHub issue #101, Thank you.

shar33f, 2022-01-12:
@x130844 does that mean you were able to build a game&watch .fw file? If so, could you please send to me, I'd love to test on my odroid-go!

shar33f, 2022-01-12:
Ok no problem and thanks for trying! Hopefully it gets merged into a future .fw release.

x130844, 2022-01-12:
@shar33f   although, feel free to try also on your side. It'd be interesting if you're getting the same errors.

closed 2022-01-12 by 32teeth