retro-esp32/RetroESP32 #102: Add support for ILI9225

h1aji, 2022-01-10:
Is there any chance to add support for ILI9225?
Its a 2.0 or 2.2 inch display 176x220

Driver here: https://github.com/Nkawu/TFT_22_ILI9225

x130844, 2022-01-10:
I'd love to add this, but can't even compile to start with. Adding support for other displays, or change pins is easy, but I think the documentation for compile from src is faulty. I asked for help, but no one seems to know.  :(

h1aji, 2022-01-11:
@x130844 works for me, I use macbook.
Followed this link https://github.com/retro-esp32/RetroESP32/blob/master/DIY.md
You just have to reinstall esp-idf and sdk

x130844, 2022-01-11:
weird,   same guide,   for me, it failed on linux and macOS at the 
"git submodule foreach git pull origin master" part. 
I tried many esp-idf and some official idf docker from espressif.

# git submodule foreach git pull origin master
Entering 'Emulators/frodo-go'
From github.com:OtherCrashOverride/frodo-go
 * branch            master     -> FETCH_HEAD
Already up to date.
Entering 'Emulators/go-play'
From github.com:OtherCrashOverride/go-play
 * branch            master     -> FETCH_HEAD
Already up to date.
Entering 'Emulators/odroid-go-handy'
From github.com:pelle7/odroid-go-handy
 * branch            master     -> FETCH_HEAD
**fatal: refusing to merge unrelated histories
fatal: run_command returned non-zero status for Emulators/odroid-go-handy**

-
I tried adding "--allow-unrelated-histories”
But failed again:
# git submodule foreach git pull origin master --allow-unrelated-histories
Entering 'Emulators/frodo-go'
From github.com:OtherCrashOverride/frodo-go
 * branch            master     -> FETCH_HEAD
Already up to date.
Entering 'Emulators/go-play'
From github.com:OtherCrashOverride/go-play
 * branch            master     -> FETCH_HEAD
Already up to date.
Entering 'Emulators/odroid-go-handy'
From github.com:pelle7/odroid-go-handy
 * branch            master     -> FETCH_HEAD
Already up to date!
Merge made by the 'recursive' strategy.
Entering 'Emulators/odroid-go-pcengine-huexpress'
**fatal: couldn't find remote ref master
fatal: run_command returned non-zero status for Emulators/odroid-go-pcengine-huexpress**

h1aji, 2022-01-11:
ok, your message is not related to ILI9225 support.
If you have source code I am happy to test it

32teeth, 2022-01-12:
*" I'd love to add this, but can't even compile to start with. Adding support for other displays, or change pins is easy, but I think the documentation for compile from src is faulty. I asked for help, but no one seems to know. :(*"

> Polite Notice
Please keep the conversation to the feature positioned, not an issue you are having elsewhere.

@h1aji 
Shouldn't be an issue, i'll see if we can get our hands on a module
In the meantime, you can add a new config for ILI9225

Take a look at [these lines](https://github.com/Nkawu/TFT_22_ILI9225/blob/master/src/TFT_22_ILI9225.cpp#L324-L395) in their source code

Just needs to be converted manually to something [like this](https://github.com/Nkawu/TFT_22_ILI9225/blob/master/src/TFT_22_ILI9225.cpp#L324-L395) - `DRAM_ATTR static const ili_init_cmd_t ili_init_cmds[]`

h1aji, 2022-01-14:
Would you be able to advise list of files that needs to be modified?
I want to give it a try. Thanks

x130844, 2022-01-14:
@h1aji  I'd say, just get it to compile first with no modifications, then once you get it to compile, we'll see about the changes needed.

h1aji, 2022-01-15:
@x130844 
This project compiles fine. If you have code for ILI9225 please paste it here and I will test. You have been given polite notice already, cant you see?

32teeth, 2022-01-15:
@h1aji 

Can you confirm that your ILI9225 is SPI or Parallel interface?
Additionally, the resolution ILI9225 is 176 x 220...   ...which will NOT display properly for most of the emulators output

Take a look at [nopnop2002](https://github.com/nopnop2002/Raspberry-ili9225spi/blob/master/ili9225.c#L215-L258) driver. *code lines are highlighted*

If you can get that running on your device.
Next steps would be to take the above [lines of code](https://github.com/nopnop2002/Raspberry-ili9225spi/blob/master/ili9225.c#L215-L258)

I will see if i can find a 9225 locally

h1aji, 2022-01-15:
I am using SPI.
Yeah, I am worried about resolution, but I wanted to give it a try to see how it looks.

> If you can get that running on your device.

I will try and will get back to you.
Thanks

h1aji, 2022-01-17:
@32teeth 
I tested the library and it works fine. 
I also found - https://github.com/Bodmer/TFT_eSPI
Tried few demos (VU meter and ping pong) and it works very nice.
[image]

Skyde72, 2022-01-29:
if you want a DIY project, check this out
https://www.instructables.com/Homemade-Odroid-go-Compatible-Game-Console/

h1aji, 2022-01-30:
@Skyde72 thanks, but I wasnt asking for instructions hof to make it. I asked for ILI9225 lcd support

h1aji, 2022-05-22:
screen is too small. not worth it

closed 2022-05-22 by h1aji