retro-esp32/RetroESP32 #103: Will esp32S3 be supported later?

FantasyGmm, 2022-01-27:
ESP32S3 ESP-IDF stable version 4.4 has been released, most features are available, SD card is not supported.
ESP32S3 more powerful than the old ESP32 , built-in PSRAM and FLASH takes up less PCB space

32teeth, 2022-01-30:
@FantasyGmm 

Yes, we are looking into the S3 (40+ pins wow), as well as the H2

FantasyGmm, 2022-02-07:
Very good, I modified the PCB and schematic of ESPlay Micro, changed the main control to ESP32, and the screen to ST7789V, which is an IPS screen, I believe the display effect will be good, I am waiting for the PCB, if there is no problem, I will upload it

pebri86, 2022-02-08:
> Very good, I modified the PCB and schematic of ESPlay Micro, changed the main control to ESP32, and the screen to ST7789V, which is an IPS screen, I believe the display effect will be good, I am waiting for the PCB, if there is no problem, I will upload it

Wow nice to hear that, don't forget to take advantage of usb feature that esp32s3 has, so the board can be useful to make an usb gadget like directly connected to pc when accessing files or copy rom.

FantasyGmm, 2022-02-09:
Yes, I use OTG and serial port, as far as I know S3 can support USB JTAG, debugging may be a lot easier

FantasyGmm, 2022-02-10:
https://oshwlab.com/FantasyGM/esplay_micro_s3,I uploaded the schematic diagram and PCB on oshwlab, I used easyEDA design, so THERE are no other documents, and I am not a professional, so if there is something wrong in the schematic diagram or PCB design, please tell me and how to correct it, thank you

SSStarku, 2024-10-31:
is the retroesp available for esp32s3?

32teeth, 2024-10-31:
You'll want to follow https://github.com/retro-esp32/RetroESP32-X project.
We will be updating