retro-esp32/RetroESP32 #104: Flashing to a ESP32 devboard

fedyfausto, 2022-01-28:
Hello guys, there are some tutorials or something else to build and run this firmware on a normal esp32?

32teeth, 2022-01-28:
> DIY

https://github.com/retro-esp32/RetroESP32/blob/master/DIY.md

pebri86, 2022-01-29:
You need devkit with esp32 wrover module or something similar with external psram to get emulator working

cctvfm, 2022-02-03:
@32teeth Could you share the .bin file to flash to an ESP32? I specifically need one for the ILI9342

deano309, 2022-12-18:
Are there other pre built firmware files available with different screen/button connections?
I have never used the ESP-IDF before but would like to change them so I can use boards which I already have (Some do not have the default pins exposed or have other hw on certain pins meaning they are unsuitable!).
I have looked at the DIY.md but am unsure how to set up the environment on Windows 10 or how to edit the files to make suitable.
My FeatherS2 is a great candidate for this project so would appreciate any further help or information in doing this.

32teeth, 2022-12-19:
> Are there other pre built firmware files available with different screen/button connections? I have never used the ESP-IDF before but would like to change them so I can use boards which I already [...]

Hey, we have moved to using the [Retro Go firmware from Ducalex](https://github.com/ducalex/retro-go). If you are looking to change the pin mapping, and flash on your own you will need to use the ESP-IDF

I can provision guidance and assistance if you would kindly share your 
* Board Layout
* Schematics

deano309, 2022-12-23:
HI 32teeth & thank you for your response.
I have decided to go down the route of modifying available hardware as I don't have the time to learn another environment and it is easier to keep the pin structure as-is so as to not risk breaking anything or having to rebuild it if there is an update!
I am more comfortable doing the hardware mods such as the one shown here: https://www.instructables.com/Homemade-Odroid-go-Compatible-Game-Console/
I am already very impressed with how this build runs on my Odroid Go and will be making a couple of consoles from old DMG cases I have in my parts box.
:)

closed 2023-03-01 by 32teeth