retro-esp32/RetroESP32 #105: Romart file not available error 404

effgarces, 2022-02-06:
The romart archive is no longer available, error 404.

Also, I can't find info on what the correct format should be. I already have romart from the go-play firmware, in .art format, but it seems to be ignored.

32teeth, 2022-02-07:
We never provided them
You need to create them yourselves

[You this tool we created](https://convert-image-to-rom-art.herokuapp.com/)

effgarces, 2022-02-07:
I was following the directions over at https://github.com/retro-esp32/RetroESP32/wiki/Romart-Usage and the romart archive referred there is not available.

As for the tool yes I already checked/used it, the main isssue is that the romart archive from the go-play firmware seems to ignored and does not load. Does the romart for the retroesp32 needs something different than that format? The images are already in the art format and in their respective folders.

closed 2022-02-14 by 32teeth