retro-esp32/RetroESP32 #108: Module support

Reubertt, 2022-06-23:
Would it be possible in a newer version to support modules as in ESPboy? I've been thinking about how to implement this function through the cartridge connector, so you could create modules in cartridge formats. This could give the Retro-ESP32 a lot more functions. There is the problem of the battery location, but maybe the location of the old battery is valid, or else using the cartridge itself as a battery. I don't have the step file to check if it's possible.

32teeth, 2022-06-24:
Hey. Thanks for the comment.
There is the [G32 ESP32](https://www.myretrogamecase.com/products/game-mini-g32-esp32-retro-gaming-kits-1-1) already that seems to support the basics of what you are looking for

Reubertt, 2022-06-24:
Physically it looks like yes, but from what I've observed the G32 has connection pins with the GPIO of ESP only for SD card. My idea was to use odroid's "10 pin" header on cartridge. Or if it's possible, the other esp's own GPIO to create RFID modules, HF and so on with the arduino function.

32teeth, 2022-07-01:
Can you help me understand better the problem you are trying to solve?
* Are you using an Odroid-Go? 
* Are you wanting to connect a cartridge directly to the Odroir-Go?

closed 2022-07-01 by 32teeth

Reubertt, 2022-07-03:
I was intending to build a board like the RetroESP Hardware to use the Arduino platform and have a certain freedom with the device. But I've already decided that I'm going to make a project derived from your board, I don't have much electronic technical knowledge but I should publish it on Github when I'm a little more advanced. Thanks in advance for what you've already done for odroid-go and I hope you'll check out the project when I publish it. ;)