retro-esp32/RetroESP32 #111: New ESP32 modules PSRAM error

giltal, 2022-12-28:
Hi
First, huge thanks for your awesome work!!!
I have cloned the Odroid-GO to create a better user experience (3.2" screen, "ATARI" like joystick, and cherry-MX buttons...
I used off the shelf TTGO ESP32 module, when I flushed the FW I got into an issue with the PSRAM, it is detected as 4095Mbit size which leads the emulator to crash, I found an old ESP32 module with an "old" version of the PSRAM, replaced them and it works like a charm.
Seems like the SDK version used to build ver 2.8 does not handle the "new" version of PSRAM chip right.
I guess if you try to assemble a new Retro ESP32 console with a newly released ESP32-Wroover module you will face the same issue.
If a version with an override for the PSRAM size detection can be generated (forcing 64Mbit size) it will solve the issue.
I can check it if needed.
Thanks in advance and happy new year guys!!!! :-)

32teeth, 2022-12-28:
This is done through your version of ESP-IDF when flashing

closed 2022-12-28 by 32teeth

giltal, 2022-12-28:
Thanks, will give it a try ...